turtlebot/turtlebot4 #2: Nav launch files are not in best practices, and will not use composition

SteveMacenski, 2022-02-28:
Please look over the current state of Nav2's launch files so that your users can use the best in Nav2 on Rolling. You do not want to copy-paste our launch files over. Instead, you want to make a launch file that launches Nav2's files with your custom configuration read-ins or additional nodes (like base, simulator, etc).

roni-kreinin, 2022-03-01:
Is there a way to run async slam using the Nav2 launch files? I find the performance is much better on the RPI4 with async.

SteveMacenski, 2022-03-01:
You can always swap out the localization launch file, but you shouldn't be overriding the navigation launch file -- only the highest level bringup which would include a localization launch include.

PR #4 (2022-03-03): Use Nav2 composition
## Description

Fixes #2 

- Replaced nav2.launch.py with modified version of https://github.com/ros-planning/navigation2/blob/main/nav2_bringup/launch/bringup_launch.py.
- This launch file can be used to launch nav2, slam sync, slam async. and localization.
- Added depot map from simulator

## Type of change

- [ ] Bug fix (non-breaking change which fixes an issue)
- [x] New feature (non-breaking change which adds functionality)

## How Has This Been Tested?

Please describe the tests that you ran to verify your changes.
Provide instructions so we can reproduce. Also list [... cut]
closed 2022-03-03 by roni-kreinin