turtlebot/turtlebot4 #10: NTP sync warning

alexswerner, 2022-07-14:
On our campus network, outgoing NTP is blocked. Ubuntu by default rejects any ntp server supplied through DHCP. This results in no time synchronization on the raspberry pi as it has no RTC with battery backup. ROS suffers from this issue as does apt.

This can be fixed by using the DHCP-provided NTP server by adding a dispatch script to NetworkManager:
```
# cat /etc/NetworkManager/dispatcher.d/10-update-timesyncd 
#!/bin/sh
set -v
echo args: $@

[ -z "$CONNECTION_UUID" ] && exit 0
INTERFACE="$1"
ACTION="$2"

case $ACTION in
up | dhcp4-change | dhcp6-change)
	[ -n "$DHCP4_NTP_SERVERS" ] || exit
	mkdir -p /etc/systemd/timesyncd.conf.d
	cat <<-THE_END >"/etc/systemd/timesyncd.conf.d/${CONNECTION_UUID}.conf"
		[Time]
		NTP=$DHCP4_NTP_SERVERS
	THE_END
	systemctl restart systemd-timesyncd.service
	;;
down)
	rm -f "/etc/systemd/timesyncd.conf.d/${CONNECTION_UUID}.conf"
	systemctl restart systemd-timesyncd.service
	;;
esac
```
This file needs to be made executable. The file is taken from somewhere else.

roni-kreinin, 2022-07-15:
Thanks, I'll add a note to the manual.

closed 2022-09-15 by roni-kreinin

teufelweich, 2022-12-20:
> Thanks, I'll add a note to the manual.

I've searched for NTP inside the manual and didn't find any note.

teufelweich, 2023-01-20:
Since my WiFi setup usually doesn't allow internet connection I used my laptop as the NTP server (through chrony).
I've added the following lines to the configs.

on my laptop (server):
__/etc/chrony/chrony.conf__ 
```bash
# Allow TurtleBots to use this chrony instance as their NTP server
allow 192.168.0.0/24
```

on the Pi/Turtlebot (client):
__/etc/chrony/chrony.conf__ 
```bash
# Laptop as source
server 192.168.0.10 iburst
```

The Create3 then syncs it's time from the Turtlebot Pi.

teufelweich, 2023-01-20:
@roni-kreinin why does contain `/etc/chrony/chrony.conf` the following line?
```bash
server 192.168.186.2 presend 0 minpoll 0 maxpoll 0 iburst  prefer trust
```

should the Pi really use the Create 3 as prefered trusted source? This conflicts with the Pi being the soruce for the Create3.