turtlebot/turtlebot4 #11: Robot does not drive far after leaving the base

alexswerner, 2022-07-15:
When I start with a docked robot, and command the robot to back out of the base through `/cmd_vel` the robot just moves back about 1 meter and then stops. It can move back into the base. I use `ros2 run teleop_twist_keyboard teleop_twist_keyboard` and buttons `i`,`,` to move the robot.

roni-kreinin, 2022-07-15:
By default the Create 3 has a backup limit because the cliff sensors are only positioned on the front half of the robot. You can override the safety settings by using the Create 3 webserver. See https://iroboteducation.github.io/create3_docs/webserver/application/ for more details.

alexswerner, 2022-07-15:
Thanks @roni-kreinin, that possiblity of the existance of any hardware protections was just confusing me.