turtlebot/turtlebot4 #12: Automatic docking

alexswerner, 2022-07-15:
Is automatic docking supposed to work? I found a service for it and tried:
```
ubuntu@uwbot-01:~$ ros2 action send_goal /dock irobot_create_msgs/action/DockServo {}
Waiting for an action server to become available...
Sending goal:
     {}

Goal was rejected.
```
That happens when the dock is visible:
```
ubuntu@uwbot-01:~$ ros2 topic echo /dock
header:
  stamp:
    sec: 1657889078
    nanosec: 962197994
  frame_id: ''
dock_visible: true
is_docked: false
```
Please let me know if this is rather an Icreate3 issue.

roni-kreinin, 2022-07-15:
The docking action is provided by the Create 3 but it should work. Does the robot do anything when you call the action?

Also can you try placing the robot on the dock and using the undock action? `ros2 action send_goal /undock irobot_create_msgs/action/Undock {}`

alexswerner, 2022-07-15:
This might be related to using the cyclonedds middleware without setting the correct middleware options: `export CYCLONEDDS_URI='<CycloneDDS><Domain><General><DontRoute>true</></></></>'
`

alexswerner, 2022-07-20:
We were no longer able to reproduce this issue later on. Docking seems to work.

closed 2022-07-20 by alexswerner