turtlebot/turtlebot4 #15: Documentation: CPU usage

alexswerner, 2022-07-20:
It would be good to state how much of the raspi CPU is available for the user. When I checked with htop (so a very subjective measurement) it seems that >60% are used by the basic services. If the turtlebot4 service is not running, this is 0%. The documentation states that running computation heavy tasks off-board is better (e.g. navigation stack). That can work, but in a classroom scenario with >10 robots the network could then become a bottleneck (to be confirmed). I wish the basic packages could just be more sensitive with their CPU usage. Also logging in over ssh and working with the command line (tab completion) on the robot is painfully slow. This is definitely lowering the level of joy one can get when working with this robot that way.

roni-kreinin, 2022-07-20:
As long as there is a subscriber for the topic, the data is being sent over the network, even if both the publisher and subscriber are on the same machine. For navigation, you have the nav2 stack publishing data, and RVIZ subscribing. For the camera, the OAK-D node is publishing and `turtlebot4_diagnostics` is subscribing. If you disable diagnostics and don't have any other nodes subscribed to the image topic, you should get improved network performance (I have not tested this though).

There are plans for using shared memory to reduce network usage but I don't believe it is implemented yet for CycloneDDS. https://answers.ros.org/question/361783/is-it-possible-to-use-shared-memory-as-ipc-on-ros2-foxy/

roni-kreinin, 2022-09-15:
Closing because of no activity. Feel free to reopen.

closed 2022-09-15 by roni-kreinin