turtlebot/turtlebot4 #17: Recommendation for running multiple robots in one room

alexswerner, 2022-07-20:
There are a couple of ways to run multiple robots (independently) in one space. For running them in one classroom we are trying to implement this scenario:

Put all PIs on one 5Ghz Wi-FI ap, all bases on one 2.4Ghz Wi-Fi ap. Connected with APs with a bridge. Have a DHCP server on this bridge with fixed IPs for the robots (e.g. uwbot-01 -> 192.168.23.100). The students then join this AP with their laptops. Internet access (documentation) is also available on this network for both students and robots.
Now of course we need to seperate the DDS domains of the different robots or use ROS namespaces. From the documentation it seems that using domains (ROS_DOMAIN_ID) is the better choice, because a) general ROS instructions can be directly applied (no namespaces) b) there will be less confusion.

We are still unsure if:
- The network will become a bottleback if students run the navigation stack and/or any perception component for each robot on their laptops. Add to the rviz for debugging. Is (Cyclone-) DDS actually streaming the data once across the network or seperately for each process? Can we reduce the transmitted data rate?
- Separation with ROS_DOMAIN_IDs works as expected, @kscottz mentioned that their could be some issues.

Is this a setup you would recommend?

roni-kreinin, 2022-07-20:
Namespacing is not yet supported but is planned. There is a [PR](https://github.com/iRobotEducation/create3_sim/pull/176) for the Create 3 sim already.

Regardless, I would still recommend using ROS_DOMAIN_ID as it will be much less confusing. I have instructions [here](https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_packages.html#installing) for installing the `robot_upstart` job with a ROS_DOMAIN_ID. Each Create 3 will also need to be set to the same ROS_DOMAIN_ID, and the remote PC will need to export the ROS_DOMAIN_ID as well. I will be adding a tutorial for this soon.

I am not sure what issues @kscottz had, but I was able to set up 8 robots on the same network without any issues.

As for the network bottleneck, this will likely depend on what your network setup is and what the turtlebots are doing. If they are all streaming video data from the cameras then there may be some issues. You can reduce data rates by reducing the camera resolution and frame rate, or turning off unused channels like depth. This can be done at [launch](https://github.com/turtlebot/turtlebot4_robot/blob/galactic/turtlebot4_bringup/launch/oakd.launch.py). For navigation applications you can reduce map resolution, update rates, and other parameters by modifying the [config](https://github.com/turtlebot/turtlebot4/tree/galactic/turtlebot4_navigation/config) files.

CycloneDDS should be optimizing data rates automatically, but you can try to adjust some of the [configurations](https://github.com/eclipse-cyclonedds/cyclonedds#run-time-configuration) if necessary.

alexswerner, 2022-07-22:
Very good, we'll set things up using ROS_DOMAIN_ID then. 

Just out of curiosity, why don't we use a [https://docs.ros.org/en/foxy/Tutorials/Advanced/Discovery-Server/Discovery-Server.html](discovery server) in this scenario? It's a pretty lame centralized old school concept, but it would probably increase reliability and decrease confusion, especially in the classroom scenario where we want isolation of the different robots.

roni-kreinin, 2022-07-22:
For ROS2 Galactic we are using CycloneDDS which does not support this. It could be an interesting option to explore as we move towards Humble which uses Fast-DDS.

kscottz, 2022-07-28:
Hi @alexswerner, sorry for the delay, Github notifications go to a different e-mail address so I don't always triage them in a timely manner. 

My concern about ROS_DOMAIN_ID comes from feedback I heard from the F1 Tenth community. In that case they had approximately two dozen ROS 2 robots in a classroom setting (and closer to a few hundred at ICRA). Similarly I have heard of issues with ROS_DOMAIN_ID when the number of robots in your office gets on the order of about 100. My warning was just to make you aware that you might be bumping into scale issues with ten robots. If you do encounter that issue please let us know as soon as possible. You're one of the first groups getting a large batch of robots and we want to make sure everything runs smoothly. 

I am not sure if you are going to be bumping into network saturation issues on your APs with ten robots used concurrently, but you're probably getting in that ballpark if the connectivity is degraded. Personally, I bought a cheap, portable, access point for my TurtleBot4 ([example](https://www.tp-link.com/us/home-networking/wifi-router/tl-wr902ac/)) so I can have a reliable connection wherever I go. This makes it a lot easier for me to do demos at remote locations, and I find that I just bridge it to my home router when I return. Just some food for thought if you plan on having students shuffling robots between homes, classrooms, labs, and the occasional outreach event.

alexswerner, 2022-07-29:
When I look at the design of the multicast discovery mechnisms I totally believe that the F1 Tenth people had a bad experience with tens of robots (likely with ten or more of topics/services each), especially over Wi-Fi. We should really look into discovery server based operations, just because it creates lot less traffic. This is not about bandwidth, because discovery traffic is small packages. I'll test this as soon as we have the 15 robots and give feedback.

@roni-kreinin What are the hurdles for switching to Fast-DDS? Basic tests from my side seem to work.

We also put the robots on a dedicated access point for the same reasons. However on a university campus 2.4 and 5 Ghz spectrums are typically quite well utilized. In our lab we try to convert the important connection to 60Ghz Wi-Fi 802.11ad/ay. That is probably not feasible due to cost reasons and the technology not being availble for that use case.

roni-kreinin, 2022-08-08:
There shouldn't be any huge hurdles with Fast-DDS. We have just been targetting CycloneDDS for release so we don't have much documentation for using Fast-DDS. You may need to do some configuration to get it working nicely (https://fast-dds.docs.eprosima.com/en/v2.3.3/fastdds/ros2/ros2_configure.html). We have seen some users run in to issues with certain packages such as nav2 (https://github.com/iRobotEducation/create3_docs/discussions/101)

alexswerner, 2022-09-01:
Just to update this ticket with the recent information. We tried to operate 15 robots at once using the setup described above and there was little to no success of having even a majority join the Wi-Fi and be operational. All errors were related to the base not connecting to the Wi-Fi correctly or disconnecting after some time. The Wi-Fi connection of the Raspberry PI is no issue, so we are implementing the work around described in #26 to continue testing.

alexswerner, 2022-09-01:
Accidentially closed this

Tanjim13, 2022-11-06:
@alexswerner, How did it go? Does the above-mentioned solution work properly? Kindly let me know. Thank in advance.

Jayden-F, 2023-02-28:
@alexswerner I'm trying to set up 12 robots, although my network seems to be struggling with only 2. Do you have any suggestions/documentation for how you got 15 working?

Jayden-F, 2023-02-28:
@roni-kreinin I will give this a try and let you know how it goes. Additionally, I would like to understand what is causing the network issue and our steps to fixing it. Would you be able to further explain or point to resources, detailing why you might use fast dds over cyclone dds for the turtlebot4.

roni-kreinin, 2023-03-01:
@Jayden-F There is a good explanation for FastDDS Discovery server [here](https://fast-dds.docs.eprosima.com/en/latest/fastdds/ros2/discovery_server/ros2_discovery_server.html). It essentially reduces the amount of traffic required for discovery, and also allows us to run the TurtleBot 4 off of just the RPi4 Wi-Fi. The Create 3 can only use 2.4GHz Wi-Fi and sometimes can have problems with range, while the RPi4 supports  5GHz and typically performs better.

closed 2023-09-21 by hilary-luo