turtlebot/turtlebot4 #22: Robot Connectivity issues

HarshalDawar, 2022-08-12:
Hello,

I am using a Turltebot4 standard along with a laptop running
 - OS: Ubuntu 20.04
 - ROS Distro: Galactic

The discussion on create3docs [https://github.com/iRobotEducation/create3_docs/discussions/142](url) is regarding the robot losing connectivity after driving some distance. 

Following the Quick start up tutorials of turtlebot4, a Laptop was used to connect(ssh) to the RPi and issue Undock/Dock actions (either via ssh or directly to the robot) and also use teleop_twist_keyboard/provided paired joystick to drive the robot. A repeatable behavior as follows was observed: 

Issue undock action -> action success -> teleop drive forward (1m to 5m, varies) -> robot loses connectivity -> robot placed on dock -> connectivity restored. 

The connectivity stays alive if robot is continuously given undock/dock commands or allowed to stay on the dock instead. It does at times lose the connection while standing on the dock as well.

During these cycles, the RPi <-> Laptop connectivity does not break.

However, when the RPi is shut down and the laptop is directly issuing action / teleop commands to the Create3 base, no such connectivity issue arise and I am able to teleoperate the robot.

Any help regarding how to proceed on diagnosing/solving this would be appreciated,

Thank you!

roni-kreinin, 2022-08-12:
Can you detail your network setup? Are you using a bridged 2.4/5Ghz router?

Also can you check the Create 3 [adapter board](https://iroboteducation.github.io/create3_docs/hw/adapter/)? The USB/BLE switch should be in the USB position.

If it is in the USB position, can you place the robot on the dock with everything turned on, call `ip a` from the RPI, and post the output here?

HarshalDawar, 2022-08-12:
Hello,

I am using my University's WiFi network; all three systems are connected to the same access point, same SSID, which supports both 2.4 and 5 GHz. Upon running `ros2 multicast send/receive` I get responses on both, Laptop and RPi.

The adapter board is in the USB position.

When on dock with everything turned on, `ip a` gives me the following:
```
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether e4:5f:01:7d:2d:56 brd ff:ff:ff:ff:ff:ff
3: usb0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 62:1e:a5:30:87:a2 brd ff:ff:ff:ff:ff:ff
    inet 192.168.186.3/24 brd 192.168.186.255 scope global usb0
       valid_lft forever preferred_lft forever
    inet6 fe80::601e:a5ff:fe30:87a2/64 scope link 
       valid_lft forever preferred_lft forever
4: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether e4:5f:01:7d:2d:57 brd ff:ff:ff:ff:ff:ff
    inet 149.201.37.77/24 brd 149.201.37.255 scope global dynamic noprefixroute wlan0
       valid_lft 3188sec preferred_lft 3188sec
    inet6 fe80::e65f:1ff:fe7d:2d57/64 scope link 
       valid_lft forever preferred_lft forever
```

roni-kreinin, 2022-08-15:
Can you copy the turtlebot4 service logs and paste them here please? Call `sudo journalctl -u turtlebot4.service -r -n 1000 --no-pager` to get the last 1000 logs.

dgel27, 2022-08-15:
Hi, I also have connectivity problem with TB4.
The case is:

I have lost connection with Create® 3 only over WiFi.
When it happens, i still can connect to RPi over WiFi, and ping Create® 3 to his usb IP.
In this case i cannot control the robot from ROS2 Galactic Desktop.
The "COMM" led is OFF.

I found root case why the Create® 3 lost WiFi connection, and i think it its firmware bug, maybe you will fix it:
The WiFi connect with Create® 3 lost in place where 2 WiFi access points with same SSID and different channel transmit with about same power. It lost and don't know to which AP to connect. If i take the robot by hand, move it 3-4 meters, it found connection, take time to battery status update.

My question is:
 **Is USB connection between RPi and Create® 3 not enough to control the robot from PC over WiFi via RPi?**

Thanks,
Dmitry Gelfond

roni-kreinin, 2022-08-15:
Currently the RPi does not route traffic from the `usb0` interface to the `wlan0` interface. This means that any ROS2 data coming from the Create 3 to the RPi over `usb0` is only visible to the RPi and not to your PC connected over WiFi. Similarly, any commands you send from your PC will only reach the RPi and not the Create 3. This is why it is currently recommended to have both the Create 3 and the RPi connected to the same network over WiFi. 

We are seeing a few people having issues with this setup so I am looking into options for routing traffic. That way users will only need to connect the RPi 4 to WiFi, and the Create 3 will only send data over `usb0`.

dgel27, 2022-08-15:
Hi Roni,
Thanks for the answer.
Maybe you could me help to setup option to route traffic.
I want build the system, in which 5G doungle connected to RPi (with additional ethernet port), connect external ethernet camera to eth port of RPi, and transfer video and TB4 controls only via 5G
Is it possible?

Thanks,
Dmitry

roni-kreinin, 2022-08-15:
@dgel27 I have created #26 to track this feature. I will test some configurations and let you know if I find a good solution for your setup.

HarshalDawar, 2022-08-15:
> `sudo journalctl -u turtlebot4.service -r -n 1000 --no-pager`

The turtlebot4 service logs as requested:
https://paste.ubuntu.com/p/zcRCgkgxn7/

roni-kreinin, 2022-08-19:
@HarshalDawar Can you try switching from CycloneDDS to Fast-DDS and see if the problem persists? On the Create 3 you can change DDS by using the [webserver](https://iroboteducation.github.io/create3_docs/webserver/application/). On the Pi and your PC you can `export RMW_IMPLEMENTATION=rmw_fastrtps_cpp`.

HarshalDawar, 2022-08-24:
I haven't had access to the robot for the past few days. I will try that out and post the update soon.

roni-kreinin, 2022-08-25:
@HarshalDawar Did you `export RMW_IMPLEMENTATION=rmw_fastrtps_cpp` after restarting your PC? If not, Galactic will default back to CycloneDDS. These issues look like you have conflicting RMW.

HarshalDawar, 2022-08-25:
I did that through my .bashrc. So it should have exported it again after the reboot as well.
Since the topics seemed to have disappeared once earlier, I did the reboot. But as of the last trial, the topics were listed, yet I was not able to receive any data over them, example, for battery status. I will cross check once with the fastrtps setup again and update if anything changes.

closed 2023-04-14 by roni-kreinin