turtlebot/turtlebot4 #23: Create3 standby/storage mode: Correct reaction to buttons.

alexswerner, 2022-08-12:
The Create3 documentation (https://iroboteducation.github.io/create3_docs/hw/face/) states:
```
Button 1
Button 1 (to the left of center) is marked with a single dot (•). If held for ten seconds, the robot will go into "standby" mode,
```
and
```
Center Button
The center button is marked with a power (⏻) icon. If held for seven seconds, the robot will go into "storage mode," disconnecting its internal battery from the robot and its payload
```

Both action do not seem to be working when the robot is on dock. That seem to be expected - as discussed in another issue (still seems broken to me). However even with the robot off the dock, button 1 did still have no effect. Maybe this was due to the internal state of the robot. When the button input had no effect, the base was displaying the spinning white light and the communication LED of the payload was off. Pressing the center button still worked to shut down the robot.

What would be helpful here is to document what the expect reaction of the robot and payload to the different button presses is with the current software state. As a bonus: Once the standby mode works, is it possible to reenable power supply to the payload throught software?

roni-kreinin, 2022-08-12:
When the white light is still spinning on the Create 3 that indicates that it is still booting. Can you place the robot in standby mode when the white light is solid?

Also when the robot is in standby mode power is still supplied to external payloads.

alexswerner, 2022-08-24:
The standby mode seem to work when the base is fully start (if it works), but I understand that it does not have the desired effect of powering down the payload. How are you charging your robots in an efficient way? We could do the following: stop lidar motor and stop ros system service on the raspberry pi. Or are there other options?

roni-kreinin, 2022-08-24:
Generally I haven't had issues with charging while the standard payload is running. Do you run into the same issue with all of your turtlebots?

If you really want to power down the entire payload you can pull out the adapter board from the Create 3.

alexswerner, 2022-09-01:
We did not test this more in depth (because of the Wi-Fi issues). I will report once more testing is done with the robots off the dock.

closed 2023-04-14 by roni-kreinin