turtlebot/turtlebot4 #25: Why can't raspberry Pie 4 automatically start the radar and camera after entering the system?

ddwy, 2022-08-15:
Why can't raspberry Pie 4 automatically start the radar and camera after entering the system?

roni-kreinin, 2022-08-15:
Can you expand further? The RPLIDAR and camera nodes should both be running on boot if you are using the standard TurtleBot 4 image.

EndlessLoops, 2022-08-16:
Is it the newest lite and standard's image in this [link](http://download.ros.org/downloads/turtlebot4/)?And how to set running the rplidar and camera on boot when I use the TB4 lite ? I found In the TB4 lite it alway run the turtlebot4.service after boot,but it can't run normally the lidar and camera.

roni-kreinin, 2022-08-16:
0.1.2 is the latest image. Can you call `sudo journalctl -u turtlebot4.service -r -n 1000 --no-pager` and send over the logs from the `turtlebot4` service?

EndlessLoops, 2022-08-17:
I will install the latest lite image and test it.

closed 2022-11-14 by roni-kreinin