turtlebot/turtlebot4 #26: Traffic routing from usb0 to wlan0

roni-kreinin, 2022-08-15:
**Describe the the feature you would like**
Configure the RPi4 to route ROS2 traffic from the Create 3 to the WiFi interface so that the Create 3 does not need to be connected to WiFi.

**Other notes**
- This will make the setup process easier and help with issues like #16, #17 and #22.
- The CycloneDDS `DontRoute` setting may no longer be necessary.

dgel27, 2022-08-17:
Hi Roni,
Since the ROS used multicast packets, can forward packets to Create 3 by this way:

1. Set static IP on RPi usb0 interface in same range as RPi wlan0 interface;
2. Set static IP on Create 3 usb0 interface in same range as usb0 and wlan0 RPi interfaces;
3. Shut down link on wlan0 Create 3 interface
So, if packets multicast it will be forwarded from wlan0 RPi to usb0 RPi and to usb0 Create 3.
If it will work, could it possible to change parameters for Create 3 board?

Do you know other method to forward packets to Create 3 ?

Also, the nub question not in the topic:
Where possible to see changes in RPi SD card published?
Do i need prepare new SD, or i can update it with apt tool?

Thanks,
Dmitry

roni-kreinin, 2022-08-17:
@dgel27 We are not able to change the static IP of `usb0` on the Create 3, and I don't think we want to. The solution I am looking at is using a multicast routing tool such as [smcroute](https://github.com/troglobit/smcroute) to route the multicast traffic from `usb0` to `wlan0`.

The latest TurtleBot 4 RPi images are available at http://download.ros.org/downloads/turtlebot4/. You can follow the instructions [here](https://turtlebot.github.io/turtlebot4-user-manual/overview/quick_start.html#install-latest-raspberry-pi-image) to update your robots SD card. Note that current TurtleBot 4's shipped with the 0.1.2 image so you may not need to update.

dgel27, 2022-08-22:
The WiFi and BT on Create3 isn't so good (maybe antennas problem, and need add external)
I just found temporary solution for moving robot without Create3 BT and WiFi.
**This is not solve problem at all, just temporary solution that help me:**
 
1. Install teleop_twist_keyboard on RPi
2. Pair Joystick with PC
3. Install on the PC keyboard emulation soft like qjoypad or joytokey
4. Map keys used in teleop_twist_keyboard to joystick
5. Connect with ssh to RPi
6. Run teleop_twist_keyboard on RPi
7. Use joystick to control robots movement 

Also it better disconnect Create3 from wifi by connecting it to temporary wifi AP, and shut down this AP.
To reach Create3 web page, better create port forwarding via usb0 to RPi with iptables

Thanks,
Dmitry

roni-kreinin, 2022-08-22:
If you just want to drive the robot from your PC then this is an option. Another option is to use [topic_tools](https://github.com/ros-tooling/topic_tools) to relay the topic on the RPi. This only works for topics, and not actions/services unfortunately. 

Another way to disconnect Create 3 WiFi completely is to perform a [factory reset](https://iroboteducation.github.io/create3_docs/webserver/about/).

dgel27, 2022-08-23:
Hi Roni,
I don't know ROS at all, but maybe it possible to do routing on CycloneDDS level?

I think it possible add [static](https://github.com/eclipse-cyclonedds/cyclonedds/blob/master/docs/manual/options.md#cycloneddsdomaindiscoverypeerspeeraddress) IP's in additional to multicast address.

Thanks!

alexswerner, 2022-08-25:
Solving this issue would make our robots run much more stable. I would guess that many issues we reported so far (e.g. base no finish "boot" state) are most likely Wi-Fi connection issues. I'm on a university campus here and our 2.4Ghz band is basically unusable. I tried to improve the reliability of the 2.4Ghz Wi-Fi connection to the base (20Mhz bands, reduce available rates) but this did not help. With this knowledge, I want to say that this bugs is essentially blocking the use of the robots here. Please let us know if there is anything we can test in this area.

alexswerner, 2022-08-26:
Just out of curiousity I investigated further if the smcroute path is feasible. So far, I could get smcroute to forward ROS2 multicast traffic successfully - with working routing between two networks. However, in the use case of multiple Turtlebots on one Wi-Fi, there are typically no routes set on the external computers to reach the internal network. Hence some NAT has to be introduced. I was also successful to SNAT the multicast packets to the raspberry pis Wi-Fi address, but the external tools (e.g. ros2 topic list) did not receive them. It is not clear to me why yet. @roni-kreinin, let me know if the details would help you here.

dgel27, 2022-08-27:
> Just out of curiousity I investigated further if the smcroute path is feasible. So far, I could get smcroute to forward ROS2 multicast traffic successfully - with working routing between two [...]

Hi @alexswerner , could you provide configurations for multicast routing setup? I have tried, see, that packets are  forwarding, but this still stay the bot uncontrolled. Thanks!

roni-kreinin, 2022-08-29:
Yes, if you could provide your configuration details that would be great.

alexswerner, 2022-08-29:
So in the working case (non-Turtlebot), there are two subnets (A=192.168.45.0/24 and B=192.168.46.0/24) connected by a gateway host running smcroute (from source). On the gateway eth0 is connected to subnet B and eth1 connect to subnet A. Host in both subnets can reach each other, there is no NAT happening on the gateway. The smcroute configuration is the following:
```
phyint eth0 enable ttl-threshold 5
phyint eth1 enable ttl-threshold 5
mgroup from eth0 group 239.255.0.1
mgroup from eth1 group 239.255.0.1
mroute from eth0 group 239.255.0.1 to eth1
mroute from eth1 group 239.255.0.1 to eth0
```
In this test case, I can create a publisher on a host in subnet A and multicast discovery messages reach hosts in subnet B. Creating a subscriber on a host in subnet B I can receive the data.

In the Turtlebot4 case I have disabled the base Wi-Fi by setting an incorrect password. On the raspberry pi the smcroute config looks like this:
```
phyint usb0 enable ttl-threshold 5
phyint wlan0 enable ttl-threshold 5
mgroup from usb0 group 239.255.0.1
mgroup from wlan0 group 239.255.0.1
mroute from usb0 group 239.255.0.1 source 192.168.186.2 to wlan0
mroute from wlan0 group 239.255.0.1 to usb0
```
With these NAT rules
```
$ iptables -t nat -L -n -v
Chain POSTROUTING (policy ACCEPT 154 packets, 62364 bytes)
 pkts bytes target     prot opt in     out     source               destination         
  117 19724 MASQUERADE  all  --  *      usb0    0.0.0.0/0            0.0.0.0/0           
  105 25326 MASQUERADE  all  --  *      wlan0   0.0.0.0/0            0.0.0.0/0
```
Somehow I have to run `conntrack -F` after starting smcroute (the NAT/connection tracking needs to learn about the multicast packet flow in the correct way. Understanding this is work in progress.

With this setup i can receive multicast packages emitted by the base, forwarded and NATed by the raspberry PI and broadcasted in the Wi-i the pi is connected to. By stopping the turtlebot4 service on the raspberry pi I make sure that only the multicast traffic of the base is there. On a workstation host the Wi-Fi the raspberry pi I get this:
```
$ sudo tcpdump -i wlo1 -n -vvv host 239.255.0.1 and host 192.168.23.101 (rpi ip)
10:04:25.308371 IP (tos 0x0, ttl 63, id 46324, offset 0, flags [DF], proto UDP (17), length 280)
    192.168.23.101.59091 > 239.255.0.1.7650: [udp sum ok] UDP, length 252
    ^^^^^^^^^^^^^^^
    correctly NATed IP (not 192.168.186.2)
```
However running `ros2 topic list` does not show the topic. The reason seems to be the following. The `ros2` client is trying to reach the IP `192.168.186.2`. I found this by running `strace -f -e trace=%net ros2 topic list --no-daemon --spin-time 30`. This shows that the multicast packages are received, but connecting via unicast to this host fails (no route). Why is it even connecting to this IP? Through further investigation into the multicast traffic with wireguard I learned that the content specifies the unicast address the DDS participant is reachable at. Even installing a route on the client would only solve half of the issue. In that case the unicast traffic could reach the base, but it would not have a route back to the workstation. Can we add a route to the base? The Create3 documentation does not say anything about that. 

Another way of fixing this is probably only possible through configuring/hacking DDS implementation. Or using IPv6. But does ROS2/Create3 support this?

roni-kreinin, 2022-08-29:
This is pretty much the setup that I had, minus the NAT rules. From what I understand, DDS uses multicast just for discovery and tries to use unicast when possible for actual data transfer. This is why we see the remote PC try to directly send data to `192.168.186.2`. I think it may be possible to add routes on the RPi to route the unicast data between `wlan0` and `usb0`, but this would likely require a static IP on the remote PC. I will test it out with these NAT rules and see if I can get that unicast data routed.

alexswerner, 2022-08-29:
I justed tried the setup without NAT. My Wi-Fi normally uses the subnet 192.168.23.0/24. I now created a route `192.168.24.0/24 via 192.168.186.3` on base via DHCP. Meaning the base Wi-Fi is on and is in the subnet 192.168.23.0/24, but gets this additional route.

On the RPI, I did this:
```
ip addr add 192.168.24.101/24 dev wlan0
echo 1 > /proc/sys/net/ipv4/ip_forward
```

Then on the client, which is also part of the Wi-Fi I removed the 192.168.23.X ip address, but added
```
sudo ip addr add 192.168.24.28/24 dev wlo1 #random ip
sudo ip route add 192.168.186.0/24 via 192.168.24.101
```

Now `ros2 topic echo /odom` gives data. This is quite a hack, and also requires adding a route on the client. Also there is still traffic between base and rpi going over the Wi-Fi. I think this can be eliminated by further modification of the DHCP configuration. Basically we could put the base on a "fake" Wi-Fi which supplied the route `default via 192.168.186.3` and a IP in an entirely unused subnet. Then all traffic (unicast) would flow through the rpi.

alexswerner, 2022-08-29:
Just this would remove the opportunity to add this route to the base. Or do you want to say that this route is then potentially unnecessary without the base Wi-Fi being configured and all traffic would flow to the raspberry pi?

roni-kreinin, 2022-08-29:
The goal is for users to not have to set up the Create 3 WiFi at all. We want the RPI4 to act as a router between the Create 3 and the users remote PC. 

Thanks to your experimentation I think I have found a working solution. Here are the steps I have taken:

On the Create 3:
- Use `rmw_fastrtps_cpp`.
- Connect to WiFi network with wrong password or perform a factory reset to disable WiFi.
- ROS2 data should only be sent over `usb0` now.

On RPi4:
- Use `rmw_fastrtps_cpp`
- `sudo apt install smcroute`
- Create `/etc/smcroute.conf` and paste the multicast settings in:
```
phyint usb0 enable ttl-threshold 5
phyint wlan0 enable ttl-threshold 5
mgroup from usb0 group 239.255.0.1
mgroup from wlan0 group 239.255.0.1
mroute from usb0 group 239.255.0.1 source 192.168.186.2 to wlan0
mroute from wlan0 group 239.255.0.1 to usb0
```

On user PC:
- Use `rmw_fastrtps_cpp`
- Add route: `sudo ip route add 192.168.186.0/24 via <RPI4 IP>`

With this setup I am able to echo Create 3 topics from both the RPi4 and remote PC, as well as publish. I tested the EStop service from my PC and it worked.

Let me know if this works for you @alexswerner.

roni-kreinin, 2022-08-29:
I have tested this with `rmw_cyclonedds_cpp` and it works as well. The `<DontRoute>true</DontRoute>` option in the cyclonedds config is still required though.

alexswerner, 2022-08-29:
I tried to implement this without a factory reset and that was not able to succeed. This issue was that even when i specified wrong Wi-Fi credentials, the Create3 remembered the old ones and re-joined the Wi-Fi. Still working on it.

alexswerner, 2022-08-29:
No success so far. Replicating your setup even `ping 192.168.186.2` from the client fails (as there is no NAT). Can you run `tcpdump` on the client e.g. like this: `sudo tcpdump -i wlo1 -n -vvv net 192.168.186.0/24` and post the output here?

roni-kreinin, 2022-08-29:
Sorry looks like I didn't notice the Create 3 connect back to WiFi which is why it looked like it worked... I'll keep looking into this.

Also I did set `ip_forward` to 1, just forgot to mention it.

alexswerner, 2022-08-30:
For now, it is no longer clear if pushing an additional route to the base via DHCP is possible as the base runs udhcpc (Busybox) which does not implement additional classless routes (121 option). Why these approaches seemed to work beforehand is beyond me right now. Moving the default gateway to 192.168.186.3 is not viable as well as the default gateway is bound to the Wi-Fi interface. Needs further investigation.
Is there a way to modify the Create3 firmware? I mean we just need to have a default route installed when no Wi-Fi is configured.

alexswerner, 2022-08-30:
I have opened a ticket for Fast-DDS if multicast-only discovery/communication setup is possible. See https://github.com/eProsima/Fast-DDS/issues/2934

roni-kreinin, 2022-08-30:
I have a test setup with 2 RPI4's like this : `(User PC) <-- WiFi --> (RPI4-1) <-- Ethernet --> (RPI4-2)`. RPI4-2 represents the Create 3 in this case. I was able to set up the Fast-DDS discovery server on RPI4-1 and successfully discovered RPI4-2 topics from the User PC. I still could not see topic data on the User PC so the unicast routing is still an issue, but at least the multicast discovery worked properly.

For the discovery server to work we need to export the discovery server ip/port on the Create 3. So if I can get unicast routing working in this setup, I think I can get iRobot to release a new firmware version that allows users to configure the discovery server through the webserver. 

Unfortunately there is no way for us to manually modify Create 3 firmware.

alexswerner, 2022-08-30:
Nice experiment. I agree that the discovery server is the way to go, but for other reasons. If you talk with iRobot my suggestion is to have a low priority default route in `/etc/network/interface`. Right now it reads
```
...
# Wired interface
auto eth0
iface eth0 inet static
    pre-up ip addr flush dev eth0 > /dev/null 2>&1 || true
    address 192.168.186.2
    netmask 255.255.255.0
```
just appending 
```
    gateway 192.168.186.3
    metric 99
```
Should give us a working system and any route added through udhcpc would have a lower (preferable) metric.

roni-kreinin, 2022-08-30:
I have managed to get the discovery server setup working properly, but, as you mentioned, it requires adding a route to the Create 3.

Here is the setup:
```
User PC (192.168.0.168) <-- WiFi --> (192.168.0.187) RPI4-1 (192.168.185.3) <-- Ethernet --> (192.168.185.2) RPI4-2
```

`User PC`:
- Configure FastRTPS to be a [super client](https://fast-dds.docs.eprosima.com/en/latest/fastdds/ros2/discovery_server/ros2_discovery_server.html#daemon-s-related-commands)
- `export ROS_DISCOVERY_SERVER="192.168.0.187:11811"`
- `sudo ip route add 192.168.185.0/24 via 192.168.0.187`

Making the User PC a super client allows ros2 cli to properly display all topics without being subscribed to them. 

`RPI4-1`:
- `export ROS_DISCOVERY_SERVER="127.0.0.1:11811"`
- `fastdds discovery -i 0`
- Enable ip forwarding

Set the discovery server to the local interface with default port. Run the discovery server with default values.
Probably will need to make this a super client too but for now it doesn't matter.

`RPI4-2`:
- `export ROS_DISCOVERY_SERVER="192.168.185.3:11811"`
- `sudo ip route add 192.168.0.0/24 via 192.168.185.3`

Set the discovery server to the `RPI4-1`'s `eth0` IP. Add a route for `192.168.0.0/24` through the `RPI4-1`'s `eth0` interface.

With this configuration I can call 
```
ros2 topic pub /cmd_vel geometry_msgs/msg/Twist "linear:
  x: 0.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 0.0" 
```
on `RPI4-2` and echo it on both `RPI4-1` and `User PC`.

The issue with this is obviously that the subnet of the Users WiFi network will vary. The next thing I want to test is what you did earlier with creating a new subnet on `RPI4-1` with a constant subnet such as `192.168.131`. In this case we would route both `User PC` and `RPI4-2` data to `192.168.131.0/24`, and then add 2 more routes on `RPI4-1` which would send that data to the correct location. This way the Create 3 can have a constant route added to the firmware, and would only require the port number of the discovery server to be configured in the webserver.

alexswerner, 2022-08-31:
Here is a work around to the problem we experience with multi-cast routing. This uses a Layer 2 VPN (openvpn) to give the client PC access to the internet usb0 network (192.168.186.0/24).

On raspberry PI
$ openvpn --genkey --secret static.key

Have `server.cfg` with
```
dev tap
ifconfig 192.168.186.9 255.255.255.0
secret static.key
```
Start VPN server
```
openvpn --config server.cfg
```
Create bridge
```
ip link add name bridge type bridge
ip link set bridge up
ip link set usb0 master bridge
ip link set tap0 master bridge
ip link set usb0 promisc on
ip link set tap0 promisc on
```

On client
$ scp RPI:static.key ./
Have file `client.cfg`
```
# raspberry pi wi-fi address
remote 192.168.23.101
dev tap
ifconfig 192.168.186.10 255.255.255.0
secret static.key
```
Start VPN client
```
sudo openvpn --config client.cfg
```

Now `ros2 topic echo /odom` should work.

Know issue: 192.168.186.3 is not reachable from the client (ARP issues)

alexswerner, 2022-10-17:
@roni-kreinin is there any development in this direction? I see there is no reaction on the ticket you opened with iRobot. For us, the tunnel setup described above is robust, but introduces one more hurdle to operate the robot. So there is no urgency, but it would definitely improve the usability (no tunnel) and range of applications (multi-robot*).

alexswerner, 2022-11-07:
After some further development, I want to share our current setup. We run our robots with disabled base Wi-Fi, and use a tunnel
to connect the client/developer PC to the internal turtlebot network at 192.168.186.0/24. I wrote down some notes here https://gist.github.com/alexswerner/cdc9b9185cd2da0eefc2c404ce405f0f

roni-kreinin, 2022-11-07:
@alexswerner Thanks for the notes. I have been discussing this issue with iRobot and they are working on adding IP routing and discovery server options to the Create 3 webserver. I am hoping that the discovery server solution is sufficient, as it should be a bit simpler to set up than the VPN.

I will comment here when I have more updates.

roni-kreinin, 2023-02-03:
With the release of G.4.1 firmware on the Create 3, we can now use the FastDDS discovery server on the TurtleBot 4. This networking configuration allows us to use the Create 3 with only the USB-C connection, meaning that only the RPi4 needs to connect to Wi-Fi.

I have written up instructions on how to set up this configuration here:

https://gist.github.com/roni-kreinin/8fbb20cb4603b8eb5e3961167fb22cd4

The instructions will be moved to the user manual once it is updated for Humble.

Reply on this issue if this configuration worked for you, or if you have any issues.