turtlebot/turtlebot4 #29: RViz2 can't show the correct  model

EndlessLoops, 2022-08-17:
I found it  always can;t show the correct model in RViz2. 

[image]

```
$ ros2 launch turtlebot4_viz view_model.launch.py
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2022-08-17-10-02-09-090179-NUC10I5-3914
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [3916]
[rviz2-1] [INFO] [1660701729.625289952] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [INFO] [1660701729.625436821] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-1] [INFO] [1660701729.650154951] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [INFO] [1660701729.858087504] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] Warning: Invalid frame ID "wheel_drop_left" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.3/src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "wheel_drop_right" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.3/src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "wheel_drop_left" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.3/src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "wheel_drop_right" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.3/src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "wheel_drop_left" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.3/src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "wheel_drop_right" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.3/src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "wheel_drop_left" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.3/src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "wheel_drop_right" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.3/src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "wheel_drop_left" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.3/src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "wheel_drop_right" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.3/src/buffer_core.cpp
```

roni-kreinin, 2022-08-17:
RVIZ is showing the robot model that is published to the `/robot_description` topic. Can you call `cat /etc/turtlebot4` on the RPi and make sure you are using the Lite image?

EndlessLoops, 2022-08-17:
I have two robot. I use the tb4-standard and tb4-lite at the same time. Tb4-standard's porblem is that RViz can't show the correct model .

roni-kreinin, 2022-08-17:
Have you configured the `ROS_DOMAIN_ID` for each robot? You cannot have both robots on the same `ROS_DOMAIN_ID` otherwise their topics will interfere.

EndlessLoops, 2022-08-17:
I did it . They have different `ROS_DOMAIN_ID `and their communication work find.

roni-kreinin, 2022-08-17:
Is the screenshot posted above when you are trying to view the Standard in RVIZ?

EndlessLoops, 2022-08-17:
Yes，it always show it .

roni-kreinin, 2022-08-17:
It is showing the correct model. Are you referring to the error messages from rviz2?

EndlessLoops, 2022-08-17:
Can't you see the Create® 3  multizone bumper separating from the robot in the above screenshot  ？

roni-kreinin, 2022-08-17:
This is just a visual bug. I already have an open ticket on the create3_sim repo (https://github.com/iRobotEducation/create3_sim/issues/170) so I will close this as duplicate.

closed 2022-08-17 by roni-kreinin