turtlebot/turtlebot4 #30: Maximum driving force of a standard Turtlebot 4

duanmujie, 2022-08-17:
Dear Turtlebot 4 supporting team,

I am a researcher from CUHK. Since I want to use the Turtlebot 4 to push an object, I want to know the maximum driving force of a standard Turtlebot 4. I only learned in the instruction document that its load is 9kg, but there is no data on the driving force. Could your team helped test the push functionality? This is very important to me, and I think relevant researchers need the same data as me. If you can answer me, I will be very grateful.

roni-kreinin, 2022-08-17:
We will do some testing and get you an answer.

apopielas, 2022-08-24:
We tested the driving force for the Turtlebot4 and measured 47.2N on carpet and 44.8N on bare floor. Hopefully that helps!

roni-kreinin, 2022-09-15:
Closing because of no activity. Feel free to reopen.

closed 2022-09-15 by roni-kreinin