turtlebot/turtlebot4 #34: Create3 boot issues

alexswerner, 2022-08-23:
Sometimes the Create3 base does not start correctly. I observe the white rotating white ring status, and later can hear the "Happy Sound". The base is connected to the Wi-Fi during parts of this process, which takes 2 to 5 minutes. After this a second boot cycle seems to start (back to rotating ring). This cycle repeats. 

Here is the relevant log file.
[create3.log](https://github.com/turtlebot/turtlebot4/files/9402305/create3.log)

Also, there seems to be no way to force power down during this boot cycle.

alexswerner, 2022-08-24:
We did switch to fast-dds and things seem to be a bit more robust. Still observing. Wi-Fi could also be an issue and we will investigate this further. 

> Also, you should be able to force a power cycle by holding the power button for 20-30s.
That should help.

alexswerner, 2022-08-24:
Further confirmation that fast-dds in general works, however the time for discovery seem to be quite long (minutes). I'm talking about the time between `ros2 topic echo /battery state` and the time the first message is printed. This could be many things, but this is the most important issue we are facing right now in the middleware area.
To address this: Is there any option to test the robot with a discovery server. On the rasberry pi this is probably feasible for us, but how do we configure this on the base?

roni-kreinin, 2022-08-24:
I have noticed that `/battery_state` won't publish for some time after a reboot, unless the actual battery state changes. Try taking the robot off the dock, or testing with another topic like `/odom`. Regarding the discovery server, I am not familiar with what is required to set it up but users don't have any configuration access on the Create 3 beyond what is available on the webserver. This is a feature I can discuss with iRobot but will likely have to wait until after their first Humble release.

alexswerner, 2022-09-01:
I believe the issues I was having where very likely all network-related. I also find that `/battery_state` is not reliably published when the robot is on the base.

closed 2022-09-01 by alexswerner