turtlebot/turtlebot4 #35: Configuration/Automated Setup

alexswerner, 2022-08-24:
We have now received our shipment of Turtlebot4 robots, and found that setup can be quite time consuming. What we apply right now to our robots:

# Step 1: Raspbery PI
- Connect to TurtleBot4 Wi-Fi
- log in via ssh (ubuntu@10.42.0.1 pw: turtlebot4) and execute
```
export BOTID=4 # example for robot no 4
install.py standard ${BOTID} # change ROS_DOMAIN_ID for services
echo export ROS_DOMAIN_ID=${BOTID} >> .bashrc # change ROS_DOMAIN_ID for shell env
echo uwbot-${BOTID} | sudo tee /etc/hostname # change hostname
sed -i "s/rmw_cyclonedds_cpp/rmw_fastrtps_cpp/" .bashrc # change middleware (also applied to services?)
echo -e \\n | sudo wifi.sh -s turtlebot -p SOMEPW -r CA # join Wi-Fi
```
# Step 2: Base
- Put base in setup mode
- Joint Wi-Fi Create-XXXX
- Use browser to modify:
  - ROS_DOMAIN_ID
  - RMW implementation - fastdds
  - set safety override full
  - set hostname
  - Wi-Fi

# Step 3: Install packages
- Connect to raspberry pi over ssh through our Wi-Fi
- Install missing middleware packages: `sudo apt install ros-galactic-rmw-fastrtps-cpp`
- Shut down robot
- Start robot again

# Step 4: Test
- `ros2 topic echo /scan`
- `ros2 topic echo /battery_state`

This is quite a lengthy process. Realistically this takes 0.5h to 1h per robot. We will not automate it right now (tried) because we are half done already, but here are some things which would simplify the process, in order of importance
- Webserver of the Create3 is very slow. Likely because the CPU of the base seems to be satured quite often (ping >5seconds). Either improve this, add a suitable (HTTP) configuration API or some other way to configure the base.
- Create3 configuration is not always successfully applied and the base does not join the Wi-Fi (may be user error)
- Rebooting the robot takes very long
- Install fast-dds packages by default
- When the Wi-Fi credentials have been entered wrong on the raspberry pi there is no easy recovery.

roni-kreinin, 2022-08-24:
- I have been working on a script that makes setting the ROS_DOMAIN_ID and RMW_IMPLEMENTATION much easier. You can test out the updated `install.py` and new `ros_config.sh` scripts from [here](https://github.com/turtlebot/turtlebot4_setup/tree/roni-kreinin/domain_id/scripts). The `ros_config.sh` script sends the ROS_DOMAIN_ID and RMW to the Create3 with a `curl` command, so the Create 3 doesn't need to be placed in AP mode. It also reinstalls the `turtlebot4` job with the updated settings and applies the changes to `.bashrc`.

- I haven't personally seen the Create 3 webserver be that slow but if you can get some logs it would be good to make an issue on iRobots [github](https://github.com/iRobotEducation/create3_docs). I think it may also be possible to forward the webserver through the RPi so that we don't have to place the robot in AP mode. I will check with iRobot.

- How long is "very long" for rebooting? The Create 3 might be struggling to connect to WiFi if its several minutes. I am trying to get the TTB4 to work with just WiFi on the RPi (#26) but it is still a WIP.

- I will be releasing a 0.1.3 image soon that will have fastrtps installed by default, along with the previously mentioned scripts.

- The WiFi script has a confirmation prompt already so I think that user error should be fairly minimal here. You can always connect an Ethernet cable and SSH in through that. See https://turtlebot.github.io/turtlebot4-user-manual/troubleshooting/faq.html#access-the-rpi-over-ethernet.

alexswerner, 2022-08-25:
Thanks. I think the suggestions address the issues. We'll keep testing.

closed 2022-08-25 by alexswerner