turtlebot/turtlebot4 #37: Raspberry pi not reachable via ethernet interface

alexswerner, 2022-08-25:
We accidentially misconfigured the Wi-Fi on one robot and now try to recover it from this state. Connecting via ethernet as described here https://turtlebot.github.io/turtlebot4-user-manual/troubleshooting/faq.html does not succeed, there is link on the interface, but the robot is not reachable via `192.168.185.3` (no ping). On other robots I see that there is no address assigned to the ethernet interface.

roni-kreinin, 2022-08-25:
Can you send the output of `cat /etc/netplan/50-cloud-init.yaml`?

alexswerner, 2022-08-25:
Here 
```
ubuntu@uwbot-11:~$ cat /etc/netplan/50-cloud-init.yaml
network: 
    version: 2 
    ethernets: 
        eth0: 
            dhcp4: true 
            optional: true 
            addresses: [192.168.185.3/24] 
        usb0: 
            dhcp4: false 
            optional: true 
            addresses: [192.168.186.3/24] 
    version: 2 
    wifis: 
        renderer: NetworkManager 
        wlan0: 
            optional: true 
            access-points: 
                turtlebot: 
                    password: tobeltrut 
                   # mode: ap 
                   # band: 5GHz 
            dhcp4: true

ubuntu@uwbot-11:~$ ip addr show
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether e4:5f:01:bb:93:5b brd ff:ff:ff:ff:ff:ff
3: usb0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 2e:93:09:d6:ea:25 brd ff:ff:ff:ff:ff:ff
    inet 192.168.186.3/24 brd 192.168.186.255 scope global usb0
       valid_lft forever preferred_lft forever
    inet6 fe80::2c93:9ff:fed6:ea25/64 scope link 
       valid_lft forever preferred_lft forever
4: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether e4:5f:01:bb:93:5e brd ff:ff:ff:ff:ff:ff
    inet 192.168.23.111/24 brd 192.168.23.255 scope global dynamic noprefixroute wlan0
       valid_lft 33613sec preferred_lft 33613sec
    inet6 fe80::e65f:1ff:febb:935e/64 scope link 
       valid_lft forever preferred_lft forever
```

And here are some relevant lines from journalctl
```
Apr 21 12:54:47 uwbot-11 cloud-init[653]: Cloud-init v. 22.2-0ubuntu1~20.04.3 running 'init' at Thu, 21 Apr 2022 12:54:46 +0000. Up 13.28 seconds.
Apr 21 12:54:47 uwbot-11 cloud-init[653]: ci-info: +++++++++++++++++++++++++++++++++++++++Net device info+++++++++++++++++++++++++++++++++++++++
Apr 21 12:54:47 uwbot-11 cloud-init[653]: ci-info: +--------+-------+-----------------------------+---------------+--------+-------------------+
Apr 21 12:54:47 uwbot-11 cloud-init[653]: ci-info: | Device |   Up  |           Address           |      Mask     | Scope  |     Hw-Address    |
Apr 21 12:54:47 uwbot-11 cloud-init[653]: ci-info: +--------+-------+-----------------------------+---------------+--------+-------------------+
Apr 21 12:54:47 uwbot-11 cloud-init[653]: ci-info: |  eth0  | False |              .              |       .       |   .    | e4:5f:01:bb:93:5b |
Apr 21 12:54:47 uwbot-11 cloud-init[653]: ci-info: |   lo   |  True |          127.0.0.1          |   255.0.0.0   |  host  |         .         |
Apr 21 12:54:47 uwbot-11 cloud-init[653]: ci-info: |   lo   |  True |           ::1/128           |       .       |  host  |         .         |
Apr 21 12:54:47 uwbot-11 cloud-init[653]: ci-info: |  usb0  |  True |        192.168.186.3        | 255.255.255.0 | global | 2e:93:09:d6:ea:25 |
Apr 21 12:54:47 uwbot-11 cloud-init[653]: ci-info: |  usb0  |  True | fe80::2c93:9ff:fed6:ea25/64 |       .       |  link  | 2e:93:09:d6:ea:25 |
Apr 21 12:54:47 uwbot-11 cloud-init[653]: ci-info: | wlan0  | False |              .              |       .       |   .    | e4:5f:01:bb:93:5e |
Apr 21 12:54:47 uwbot-11 cloud-init[653]: ci-info: +--------+-------+-----------------------------+---------------+--------+-------------------+
Apr 21 12:54:47 uwbot-11 cloud-init[653]: ci-info: ++++++++++++++++++++++++++++Route IPv4 info++++++++++++++++++++++++++++
Apr 21 12:54:47 uwbot-11 cloud-init[653]: ci-info: +-------+---------------+---------+---------------+-----------+-------+
Apr 21 12:54:47 uwbot-11 cloud-init[653]: ci-info: | Route |  Destination  | Gateway |    Genmask    | Interface | Flags |
Apr 21 12:54:47 uwbot-11 cloud-init[653]: ci-info: +-------+---------------+---------+---------------+-----------+-------+
Apr 21 12:54:47 uwbot-11 cloud-init[653]: ci-info: |   0   | 192.168.186.0 | 0.0.0.0 | 255.255.255.0 |    usb0   |   U   |
Apr 21 12:54:47 uwbot-11 cloud-init[653]: ci-info: +-------+---------------+---------+---------------+-----------+-------+
Apr 21 12:54:47 uwbot-11 cloud-init[653]: ci-info: +++++++++++++++++++Route IPv6 info+++++++++++++++++++
Apr 21 12:54:47 uwbot-11 cloud-init[653]: ci-info: +-------+-------------+---------+-----------+-------+
Apr 21 12:54:47 uwbot-11 cloud-init[653]: ci-info: | Route | Destination | Gateway | Interface | Flags |
Apr 21 12:54:47 uwbot-11 cloud-init[653]: ci-info: +-------+-------------+---------+-----------+-------+
Apr 21 12:54:47 uwbot-11 cloud-init[653]: ci-info: |   1   |  fe80::/64  |    ::   |    usb0   |   U   |
Apr 21 12:54:47 uwbot-11 cloud-init[653]: ci-info: |   3   |  multicast  |    ::   |    usb0   |   U   |
Apr 21 12:54:47 uwbot-11 cloud-init[653]: ci-info: +-------+-------------+---------+-----------+-------+
Apr 21 12:54:48 uwbot-11 systemd[1]: Finished Initial cloud-init job (metadata service crawler).
Apr 21 12:54:48 uwbot-11 systemd[1]: Reached target Cloud-config availability.
```

Hope this helps.

alexswerner, 2022-08-26:
No this was another robot without the cable connected. We did not find the time to open up the bricked robot yet.

alexswerner, 2022-09-06:
With another robot I could now successfully connect via Ethernet. Hence, the configuration seems to be correct. I am suspecting that the SD card / filesystem of the raspberry pi was corrupted as reflashing that helped. That latter issue I have seen with multiple robots. The display does not light up again. I do not have access to a micro-HDMI adapter so further investigation will have to wait.

closed 2022-11-14 by roni-kreinin