turtlebot/turtlebot4 #39: Cannot ssh into Raspberry Pi of turtlebot

Tanjim13, 2022-08-30:
I am working with turtlebot4. I was setting my connection with my lab's wifi SSID. After using this command "sudo wifi.sh -s '<WIFI_SSID>' -p '<WIFI_PASSWORD>' -r <REGULATORY_DOMAIN> && sudo reboot" my turtlebot4's IP address did not upgraded on the LED screen. I connected the create3 with the router. Then I restarted the turtlebot4. Now the led screen is black. And I cannot ssh into the raspberry pi. I need a solution. Could you kindly help me?

roni-kreinin, 2022-08-30:
@Tanjim13 Can you try to SSH into the Pi over ethernet? Instructions [here](https://turtlebot.github.io/turtlebot4-user-manual/troubleshooting/faq.html#access-the-rpi-over-ethernet).

Tanjim13, 2022-08-30:
@roni-kreinin Thank you so much. It worked.

closed 2022-09-01 by roni-kreinin