turtlebot/turtlebot4 #42: I cannot connect the Raspberry using ethernet

nidia-bucarelli, 2022-09-01:
I am trying to SSH into the Pi over ethernet using the instructions from this link: https://turtlebot.github.io/turtlebot4-user-manual/troubleshooting/faq.html#access-the-rpi-over-ethernet. I am using Windows. I was able to connect the Pi through an adapter (ethernet - USB). However, I cannot set the manual connection. I get an error. See images below. 

[image]
[image]

I am trying to use this approach because after doing ssh ubuntu@10.42.1 and trying to connect to the university WIFI, the AP is not visible anymore and I cannot see the device connected when i do `ros2 topic echo /ip`. The screen is also off/black. The robot is ON.

roni-kreinin, 2022-09-01:
Can you scroll down to show the rest of the settings?

Also you want to set your IP address on the `192.168.185` subnet, and set the gateway to `192.168.185.1`.

If its possible to set DNS to automatic, do that.

nidia-bucarelli, 2022-09-01:
I cannot make DNS automatic.
I changed the settings as instructed, still getting an error.
[image]

There are no many additional settings:
[image]

nidia-bucarelli, 2022-09-01:
I was able to get connected to the PI using ssh ubuntu@192.168.185.3. 
But the Pi does not want to connect to the Internet. 

I tried an open network and my phone as a Hostpost. Nothing is working. 

[image]

[image]

roni-kreinin, 2022-09-01:
Can you try using single quotation marks around the ssid and password? i.e. `sudo wifi.sh -s 'Nidia Bucarelli (2)' `.

nidia-bucarelli, 2022-09-01:
I just tried that using my phone as a Host. It seems to be working. The screen is now ON and I can see the ip address. 

But I cannot make it work with the open network.  
I am using  `sudo wifi.sh -s 'IllinoistNet_Guest' -r US && sudo reboot`  and I am not getting a connection. Any feedback?

roni-kreinin, 2022-09-01:
The `wifi.sh` script does not currently account for open networks. I will make a change to fix that.

Meanwhile you can manually edit the netplan configuration.

`sudo nano /etc/netplan/50-cloud-init.yaml`

Under the `wifis` section you should see something like:

```
access-points:
    "IllinoistNet_Guest":
        password: ""
```

You will want to delete the `password: ""` line, and add `{}` after the SSID like this:

```
access-points:
    "IllinoistNet_Guest": {}
```

Save those changes and then call `sudo netplan apply && sudo reboot`.

Let me know if that works for you.

nidia-bucarelli, 2022-09-01:
Thank you!
I changed the file. I saved it as instructed. It still does not work. I have pasted some pictures below.
Any feedback?  

[image]

[image]

[image]

nidia-bucarelli, 2022-09-01:
I also tried to use a WIFI from a router using `sudo wifi.sh -s 'NidiaRouter' -p '12345678' -r  US && sudo reboot` . It did not want to work either. The screen is still ON, and it shows the IP address that is shown when I use my phone, but it is not connected.

So far, I can get internet only if i use my hostpost.

nidia-bucarelli, 2022-09-07:
I solved this issue. Thanks!

closed 2022-09-07 by nidia-bucarelli