turtlebot/turtlebot4 #44: I cannot see the robot in rviz.

nidia-bucarelli, 2022-09-07:
Hi!
I am connected to my robot. As instructed, I ran `ros2 launch turtlebot4_viz view_model.launch.py`.  I opened a new terminal and I ran `ros2 launch turtlebot4_viz view_robot.launch.py`.

I cannot see the robot n rviz. 
I also get the Info below about stereo and the ParticleCloud plugging.

Any feedback? thank you!

`INFO] [launch]: All log files can be found below /home/nidia/.ros/log/2022-09-06-22-57-12-135174-nidia-UX303LAB-133960
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [133973]
[rviz2-1] [INFO] [1662523032.716439940] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [INFO] [1662523032.716620006] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-1] [INFO] [1662523032.744822018] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [ERROR] [1662523032.837510857] [rviz2]: PluginlibFactory: The plugin for class 'nav2_rviz_plugins/ParticleCloud' failed to load. Error: According to the loaded plugin descriptions the class nav2_rviz_plugins/ParticleCloud with base class type rviz_common::Display does not exist. Declared types are  rviz_default_plugins/Axes rviz_default_plugins/Camera rviz_default_plugins/FluidPressure rviz_default_plugins/Grid rviz_default_plugins/GridCells rviz_default_plugins/Illuminance rviz_default_plugins/Image rviz_default_plugins/InteractiveMarkers rviz_default_plugins/LaserScan rviz_default_plugins/Map rviz_default_plugins/Marker rviz_default_plugins/MarkerArray rviz_default_plugins/Odometry rviz_default_plugins/Path rviz_default_plugins/PointCloud rviz_default_plugins/PointCloud2 rviz_default_plugins/PointStamped rviz_default_plugins/Polygon rviz_default_plugins/Pose rviz_default_plugins/PoseArray rviz_default_plugins/PoseWithCovariance rviz_default_plugins/Range rviz_default_plugins/RelativeHumidity rviz_default_plugins/RobotModel rviz_default_plugins/TF rviz_default_plugins/Temperature rviz_default_plugins/Wrench
[rviz2-1] [ERROR] [1662523032.873505590] [rviz2]: PluginlibFactory: The plugin for class 'nav2_rviz_plugins/GoalTool' failed to load. Error: According to the loaded plugin descriptions the class nav2_rviz_plugins/GoalTool with base class type rviz_common::Tool does not exist. Declared types are  rviz_default_plugins/FocusCamera rviz_default_plugins/Interact rviz_default_plugins/Measure rviz_default_plugins/MoveCamera rviz_default_plugins/PublishPoint rviz_default_plugins/Select rviz_default_plugins/SetGoal rviz_default_plugins/SetInitialPose
[rviz2-1] [ERROR] [1662523032.892963483] [rviz2]: PluginlibFactory: The plugin for class 'nav2_rviz_plugins/Navigation 2' failed to load. Error: According to the loaded plugin descriptions the class nav2_rviz_plugins/Navigation 2 with base class type rviz_common::Panel does not exist. Declared types are 
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[rviz2-1] [INFO] [1662523462.098945648] [rclcpp]: signal_handler(signal_value=2)

`

roni-kreinin, 2022-09-12:
The stereo message can be ignored.

Are you able to see all of the turtlebot4 topics in the terminal that you are running `view_robot.launch.py` from?

nidia-bucarelli, 2022-09-12:
When I run view_robot.launch.py this is the only message that I can see. I don't see the TB.

```
'nidia@nidia-UX303LAB:~$ source /opt/ros/galactic/setup.bash
nidia@nidia-UX303LAB:~$ ros2 launch turtlebot4_viz view_robot.launch.py
[INFO] [launch]: All log files can be found below /home/nidia/.ros/log/2022-09-12-11-09-54-355715-nidia-UX303LAB-5354
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [5356]
[rviz2-1] [INFO] [1662998995.217735202] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [INFO] [1662998995.217951583] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-1] [INFO] [1662998995.259768217] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [WARN] [1662998995.523728962] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[rviz2-1] [WARN] [1662998995.527474476] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[rviz2-1] [WARN] [1662998995.530945415] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
```

roni-kreinin, 2022-09-12:
Can you run `ros2 topic list` in that terminal?

nidia-bucarelli, 2022-09-12:
This is how it looks like:

`nidia@nidia-UX303LAB:~$ source /opt/ros/galactic/setup.bash
nidia@nidia-UX303LAB:~$ ros2 topic list
/clicked_point
/downsampled_costmap
/downsampled_costmap_updates
/global_costmap/costmap
/global_costmap/costmap_updates
/global_costmap/voxel_marked_cloud
/initialpose
/local_costmap/costmap
/local_costmap/costmap_updates
/local_costmap/published_footprint
/local_costmap/voxel_marked_cloud
/local_plan
/map
/map_updates
/mobile_base/sensors/bumper_pointcloud
/parameter_events
/particle_cloud
/plan
/robot_description
/rosout
/scan
/tf
/tf_static
/waypoints
`

roni-kreinin, 2022-09-12:
It looks like none of the turtlebot4 topics are showing. Can you make sure the robot is properly connected and on the same `ROS_DOMAIN_ID`?

nidia-bucarelli, 2022-09-13:
The robot is connected. I am using Desktop Ubuntu, and I have the ROS_DOMAIN_ID set to 1. For the TB4, it also has been set to 1 (See below). When I run `ros2 launch turtlebot4_navigation slam_sync.launch.py` and then I run ros2 node list (using a new terminal) TB4 does not show up.

`ubuntu@ubuntu:~$ printenv | grep -i ROS
ROS_VERSION=2
ROS_PYTHON_VERSION=3
ROS_DOMAIN_ID=1
AMENT_PREFIX_PATH=/opt/ros/galactic
PYTHONPATH=/opt/ros/galactic/lib/python3.8/site-packages
LD_LIBRARY_PATH=/opt/ros/galactic/opt/yaml_cpp_vendor/lib:/opt/ros/galactic/opt/rviz_ogre_vendor/lib:/opt/ros/galactic/lib/aarch64-linux-gnu:/opt/ros/galactic/lib
ROS_LOCALHOST_ONLY=0
PATH=/opt/ros/galactic/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
ROS_DISTRO=galactic
`

`nidia@nidia-UX303LAB:~$ printenv | grep -i ROS
ROS_VERSION=2
ROS_PYTHON_VERSION=3
ROS_DOMAIN_ID=1
AMENT_PREFIX_PATH=/opt/ros/galactic
PYTHONPATH=/opt/ros/galactic/lib/python3.8/site-packages
LD_LIBRARY_PATH=/opt/ros/galactic/opt/yaml_cpp_vendor/lib:/opt/ros/galactic/opt/rviz_ogre_vendor/lib:/opt/ros/galactic/lib/x86_64-linux-gnu:/opt/ros/galactic/lib
ROS_LOCALHOST_ONLY=0
PATH=/opt/ros/galactic/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
ROS_DISTRO=galactic`

roni-kreinin, 2022-09-13:
The `ROS_DOMAIN_ID` needs to also be set when installing the `robot_upstart` job that runs turtlebot4 nodes in the background. Try running `uninstall.py` on the TB4 to uninstall the job, then run `install.py standard 1` to reinstall the job for a standard TB4 with ROS_DOMAIN_ID=1.

nidia-bucarelli, 2022-09-13:
I did that. It still does not work.

roni-kreinin, 2022-09-13:
Can you call `ros2 topic list` on the RPI and make sure that the topics are showing?

nidia-bucarelli, 2022-09-13:
Below are my topics. Am I missing something? I am new user in ROS and Ubuntu.

```
ubuntu@ubuntu:~$ ros2 topic list
/battery_state
/cmd_vel
/color/preview/image
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/dock
/hazard_detection
/hmi/buttons
/hmi/display
/hmi/display/message
/hmi/led
/imu
/interface_buttons
/ip
/joint_states
/joy
/joy/set_feedback
/mouse
/parameter_events
/robot_description
/rosout
/scan
/stereo/depth
/tf
/tf_static
/wheel_status
`

roni-kreinin, 2022-09-13:
This looks correct for the TB4 nodes, however the Create 3 topics seem to be missing. Make sure you have set the ROS_DOMAIN_ID to 1 on the Create 3 as well by using the webserver.

nidia-bucarelli, 2022-09-13:
I changed the ROS_DOMAIN_ID to 1 on the Create 3.
The list of topics on the RPI is still the same. 

Below is is the list of topics when I run `ros2 topic list `on my computer.

nidia@nidia-UX303LAB:~$ ros2 topic list
/battery_state
/cmd_audio
/cmd_lightring
/cmd_vel
/dock
/hazard_detection
/imu
/interface_buttons
/ir_intensity
/ir_opcode
/kidnap_status
/mouse
/odom
/parameter_events
/robot_state/transition_event
/rosout
/slip_status
/static_transform/transition_event
/stop_status
/system_monitor/transition_event
/tf
/tf_static
/wheel_status
/wheel_ticks
/wheel_vels

roni-kreinin, 2022-09-13:
Are your 5Ghz and 2.4Ghz networks bridged? It looks like your computer and the Create 3 are on the same network, while the RPi is on a different network. 

If you are unable to bridge the 2.4/5Ghz networks then you can try just connecting the RPi4 to the 2.4GHz network that the Create 3 is on. You will notice reduced performance with streaming the camera data in this case.

nidia-bucarelli, 2022-09-14:
I connected my RPI, computer and the iCreate to a personal router. I can see my robot in Rviz, and i was able to create my map. 

However, when I connect the RPI, computer and iCreate to an open network (e.g., university network) I cannot see my robot in Rviz. Do you have any idea why this is happening? I need to be able to use the university network so I can move the robot around.

roni-kreinin, 2022-09-14:
Discovery is done with multicasting, so it might be that the university network is blocking it.

Tanjim13, 2022-09-15:
@nidia-bucarelli,  what is the ip adress of the turtlebot4, when you are connected to university's wifi network?

nidia-bucarelli, 2022-09-15:
@Tanjim13 
When I get connected to the university network, the ipd address for the RPI and iCreate are below, respectively. 
172.16.99.91  
172.16.99.164

nidia-bucarelli, 2022-09-15:
@roni-kreinin is that mdns? Does it have a “ptr name” or “service id?

dgel27, 2022-09-15:
@nidia-bucarelli The open networks usually created for guests, and main security parameters of it  is client isolation. So each connected client could reach internet, but without connections between connected clients. you can check it with 2 laptops connected to WiFi, if then have ping between them

roni-kreinin, 2022-09-15:
@nidia-bucarelli mdns is a protocol for resolving hostnames for IP addresses. For example it would let your RPi know that `172.16.99.164` is the Create 3. If that is blocked then it may cause issues, although I am not familiar with mdns.

As @dgel27  suggested, can you try pinging both devices from your computer?

nidia-bucarelli, 2022-09-22:
Hi, 
We have some additional information that we wish to seek advice on while working with our campus IT staff. In doing some further troubleshooting, we have discovered a few more technical details on how the Turtle Bot unit is physically connected and understanding how it connects the various components together.
We see that there are two “units” a Raspberry PI as well as the iRobot Create. Both have separate wireless cards and both have successfully connected to the campus Wireless Network. Both have separate IP addresses and can see each other on the network. We also saw in the configuration that there is a USB cable that connects the Raspberry PI and the iRobot Create. In the configuration the USB port and connection has a 192.168.x.x IP address. 
We have discovered that when we connect the Turtlebot to a private wireless router, and not the campus wireless router. It will work. However, when we connect to the campus network, it does not. Our suspicions are that when the Turtlebot is connected to the private router with limited security settings (as well as routing a valid 192.168.x.x address range), the network traffic as well as the USB traffic will eventually figure out where it is destined to. However, since the campus does not route 192.168.x.x it fails the connection. To justify this observation, the campus wireless controllers are registering the 192.168.x.x IP address in addition to the valid campus IP address. We also observed in the iRobot logs that there is a failure to connect to another 192.168.x.x which is suspiciously close, and our assumption is the other side of the USB connection. It appears to us, that when the robot is connected to the campus network, the USB traffic gets “confused” and sent out the wireless card instead of the physical USB cable. 
 
**Our first question, is if this assumption is correct? What would happen if we removed the network configuration of the USB connection and let the Operating system discovery the two units via Serial bus? We also noticed a NOROUTE option in the irobot site and wonder if that might solve this issue as well?
 
**Second question, looking at the documentation of the iRobot site (https://iroboteducation.github.io/create3_docs/setup/xml-config/) We want to disable Multicast and allow only unicast. Are these commands simply uploaded in the Webpage of the robot or does other special configuration need to take place on the raspberry pi?
 
Thank you!

roni-kreinin, 2022-09-23:
The `usb0` network interface uses a static subnet of `192.168.186`. The IP address of the Create 3 is `192.168.186.2` and the RPi is `192.168.186.3`. These IP addresses are static and should not be changed. I would not recommend disabling the USB connection as it is also used for syncing the RPi and Create 3 clocks. I'm also not sure what you mean by "let the Operating system discovery the two units via Serial bus".

Regarding the issues with your campus network, its likely that the security settings are blocking multicast which is used for discovery between ROS2 devices. You can try applying the configurations from iRobots documents to use unicast instead, although I have not tested these. I believe these will need to be applied on the RPi, and possibly on your PC as well. You will need to specify each peer that will be used in your network. I would recommend checking the [cyclonedds github](https://github.com/eclipse-cyclonedds/cyclonedds) for more details.

nidia-bucarelli, 2022-11-16:
Hi, I am trying to disable the multicast. I added the script below to the pi4 and the computer. This has not fixed the issue. I cannot see the topics when i run ros2 topic list from my computer. Do you have any feedback or suggestions on how to fix this issue?

XML file for my pi4:
The second IP address is my robot (pi4), the first IP address is my machine/computer.

```
 <CycloneDDS>
  <Domain>
    <Id>1</Id>
    <General>
      <NetworkInterfaceAddress>wlan0,usb0</NetworkInterfaceAddress>
      <AllowMulticast>false</AllowMulticast>
      <EnableMulticastLoopback>true</EnableMulticastLoopback>
    </General>
    <Discovery>
      <ParticipantIndex>0</ParticipantIndex>
      <Peers>
	      	<Peer Address=172.16.99.91:7410/>
           	<Peer Address=172.16.107.67:7410/>
      </Peers>
    </Discovery>
  </Domain>
</CycloneDDS>
```
XML file for my computer:
```
<CycloneDDS>
  <Domain>
    <Id>1</Id>
    <General>
      <NetworkInterfaceAddress>wlp2s0</NetworkInterfaceAddress>
      <AllowMulticast>false</AllowMulticast>
      <EnableMulticastLoopback>true</EnableMulticastLoopback>
    </General>
    <Discovery>
      <ParticipantIndex>0</ParticipantIndex>
      <Peers>
        <Peer Address="172.16.107.67:7410"/>
      </Peers>
    </Discovery>
  </Domain>
</CycloneDDS>
```

Tanjim13, 2022-11-17:
Hi @nidia-bucarelli, Can you clarify for me the router you are using in which mode? (1. Access point mode? 2. Router mode?) If it is not in router mode, kindly switch it to router mode. Cause I had to face a problem regarding this. I hope you are not facing a similar problem.

nidia-bucarelli, 2022-11-21:
Hi @Tanjim13, I do not understand your question. I am trying to get connected to the University WIFI network.
I disabled multicast on the PI4 and computer using the code below. I ran the talker message on my PI, and my computer was able to see it when I ran the listener. However, the robot topics do not show up on my computer when I run ros2 topic list.

I am not sure what else to do?

``
My computer:

<CycloneDDS>
  <Domain>
    <Id>1</Id>
    <General>
      <NetworkInterfaceAddress>wlp59s0</NetworkInterfaceAddress>
      <AllowMulticast>false</AllowMulticast>
      <EnableMulticastLoopback>true</EnableMulticastLoopback>
    </General>
    <Discovery>
      <ParticipantIndex>0</ParticipantIndex>
      <Peers>
        <Peer Address="172.16.107.67"/> 
        <Peer Address="172.16.99.91"/> 
      </Peers>
    </Discovery>
  </Domain>
</CycloneDDS>
Robot configuration:

<CycloneDDS>
  <Domain>
    <Id>1</Id>
    <General>
      <NetworkInterfaceAddress>wlan0,usb0</NetworkInterfaceAddress>
      <AllowMulticast>false</AllowMulticast>
      <EnableMulticastLoopback>true</EnableMulticastLoopback>
    </General>
    <Discovery>
      <ParticipantIndex>0</ParticipantIndex>
      <Peers>
          <Peer Address="localhost"/> 
          <Peer Address="172.16.99.91"/> 
      </Peers>
    </Discovery>
  </Domain>
</CycloneDDS>

``

roni-kreinin, 2022-11-21:
@nidia-bucarelli Can you try adding the Create 3 as a peer too?

nidia-bucarelli, 2022-11-28:
I tried to add the Create to both the computer and Pi4. It does not work. 
Do you know how I can ssh the Create? I wonder if I also need to add the file to the Create. 

Thanks!

nidia-bucarelli, 2022-11-30:
@Tanjim13  hello!
I am using the University WIFI network, not a router. I have tried to enable unicast. I still cannot see the topics. How do I change it to access point?

roni-kreinin, 2023-03-06:
@nidia-bucarelli Is the issue still persisting? The discovery server is now officially supported in Humble and should resolve multicast issues. See https://turtlebot.github.io/turtlebot4-user-manual/setup/networking.html#discovery-server for details.

closed 2023-04-14 by roni-kreinin