turtlebot/turtlebot4 #46: OAK-D Pro turtlebot4 standard Not working

Tanjim13, 2022-09-07:
- OS: (e.g. Ubuntu 20.04)
 - ROS Distro: (e.g. Galactic)
 - Turtlebot4 Standard
I ssh into raspberry pi and run this.

"ros2 launch turtlebot4_bringup oakd.launch.py"

I get a warning and an error.

[rgb_stereo_node-1] [2022-09-07 16:11:29.846] [warning] Insufficient permissions to communicate with X_LINK_BOOTED device having name "2.1". Make sure udev rules are set
[rgb_stereo_node-1] terminate called after throwing an instance of 'std::runtime_error'
[rgb_stereo_node-1]   what():  No available devices
[ERROR] [rgb_stereo_node-1]: process has died [pid 15574, exit code -6, cmd '/opt/ros/galactic/lib/depthai_examples/rgb_stereo_node --ros-args --params-file /tmp/launch_params__xstx0n9 --params-file /tmp/launch_params_95x5npru --params-file /tmp/launch_params_bl1yrtee --params-file /tmp/launch_params_m6psmrja --params-file /tmp/launch_params_jcyut_2l --params-file /tmp/launch_params_przv_h35 --params-file /tmp/launch_params_tctrfq14 --params-file /tmp/launch_params_p3q233zu --params-file /tmp/launch_params_ejkeeysl --params-file /tmp/launch_params_9eiji29l --params-file /tmp/launch_params_dt81g1j9 --params-file /tmp/launch_params_z7hen4af --params-file /tmp/launch_params_7azzg2u0 --params-file /tmp/launch_params_bd1rtao6'].

Tanjim13, 2022-09-12:
I have successfully connected it.

closed 2022-09-12 by Tanjim13

sudhanshuskr, 2023-03-09:
I am getting the same error, how did you resolve it?