turtlebot/turtlebot4 #48: `ros2 launch turtlebot4_viz view_model.launch.py` makes ros2 nodes disappear

Kohl-Stefan, 2022-09-16:
OS: Ubuntu 20.04
ROS Distro: Galactic
Installed: ros-galactic-turtlebot4-viz
Package version: 0.1.0-1focal.20220730.081410 

When launching the `ros2 launch turtlebot4_viz view_model.launch.py`, we would like to see the Rviz overview and expect the other turtlebot nodes continuously to work.

The Rviz graphic appears, but the turtlebot nodes disappear. From there on they either come and go or they do not come back anymore. Also the camera image is rarely or not updated any more.
Apart from that I am not sure how to access the the stereo camera data or if I have to launch separately special nodes for that? So far I could only reach the data in form of a single camera image via the /color/preview/image topic.  

**To Reproduce**
Provide the steps to reproduce:

1. run `ros2 launch turtlebot4_viz view_model.launch.py`
[image]
[image]

roni-kreinin, 2022-09-19:
@Kohl-Stefan Can you try calling `ros2 node list --no-daemon` and see if the nodes show up after rviz is launched? This looks like more of an issue with ros2 cli.

For viewing the camera image you may find better results using a separate application. Try running `ros2 run rqt_image_view rqt_image_view`.

The default OAK-D node should be publishing both the color preview image, and the depth image. If you want to view stereo images you may want to try running one of the DepthAI examples such as https://github.com/luxonis/depthai-ros-examples/blob/main/depthai_examples/launch/stereo.launch.py

Kohl-Stefan, 2022-10-10:
@roni-kreinin  WIth `ros2  node list --no-daemon` it is the same - the nodes do not appear anymore until I stop the command. By the way this also happens if I access the camera data via `ros2 run rqt_image_view rqt_image_view`. Do you have any idea?

closed 2023-09-21 by hilary-luo