turtlebot/turtlebot4 #50: TurtleBot4 Wifi troubleshooting

Amarok-shd, 2022-09-20:
TurtleBot4 lite while setting it up with our internal wifi setting with the help of the turtlebot4 manual.

While setting up the WIFI Setup, After SSH into the Raspberry Pi to configure its WiFi.and while configuring the Raspberry Pi's WiFi.

I am unsure if I gave the correct WIFI_SSID or password for our WIFI setting. after which I am unable to check the /ip topic for the new address. and unable to find the IP address and unable to ssh back into the robot with it.

Even with the help of nmap -sP 192.168.1.0/24 we are unable to ssh back into the robot

So, is there a way to get the Turtelbot 4 lite to its API mode? and also unable to operate using ROS2

We request for your assistance

roni-kreinin, 2022-09-20:
Try connecting to the Pi over ethernet as described here https://turtlebot.github.io/turtlebot4-user-manual/troubleshooting/faq.html#access-the-rpi-over-ethernet

Amarok-shd, 2022-09-20:
Using the ethernet interface, 192.168.185.3 I was able to ping with the turtlebot4 lite. Using the Win 10
and using PuTYy I am able to start the command window, but I don't know what is the login and password criteria.

Can you help me move forward?

roni-kreinin, 2022-09-20:
It is the same SSH credentials. Username: `ubuntu` Password `turtlebot4`.

Amarok-shd, 2022-09-20:
So, after this how should I SSH back of Turtlebot API

roni-kreinin, 2022-09-20:
Once you have entered the credentials you should be SSH'd into the Turtlebot4. Then you can run the `wifi.sh` script again to set up WiFi.

Amarok-shd, 2022-09-21:
Thank you sir, i was able to go back to Turtlebot4 API

I will close this issue.

closed 2022-09-21 by Amarok-shd