turtlebot/turtlebot4 #51: ros2 launch depthai_examples yolov4_publisher.launch.py not publishing topics

Tanjim13, 2022-09-21:
I am having a problem with this script. When I echo topics of this script it is not showing me any data on the terminal. I have tried to echo "/color/yolov4_Spatial_detections, /color/image, /stereo/depth" topics which should be published according to the script. Even I do not see any errors.
[image]

roni-kreinin, 2022-09-21:
I am seeing the same thing. This is an issue with the `depthai_examples` package though, so can you move this issue to https://github.com/luxonis/depthai-ros-examples?

Tanjim13, 2022-09-23:
I have already posted it in their forum.

roni-kreinin, 2022-09-27:
Issue discussed at https://github.com/luxonis/depthai-ros/issues/140

closed 2022-09-27 by roni-kreinin