turtlebot/turtlebot4 #52: Wifi connection

catalinamurray, 2022-09-21:
Hi, I have the Turtlebot 4 Standard and I am running it on, 
 - OS: Ubuntu 20.04
 - ROS Distro:  Galactic

I am having trouble connecting the rasberry pi and connect 3 to wifi. I keep getting the error "activation of network connection failed". 

I am also having trouble powering the Turtlebot 4 standard off, I push the power button and it flashes 3 times but it won't make the chime or power off for 30 min or more. When I do turn it on, by placing it on the doc only 3 of the leds will turn on.

roni-kreinin, 2022-09-22:
Where are you seeing this network error?

To power cycle the robot try pressing and holding the power button for 20-30s until the lightring turns off.

roni-kreinin, 2022-11-14:
Closing due to inactivity, feel free to re-open if the issue persists.

closed 2022-11-14 by roni-kreinin