turtlebot/turtlebot4 #53: Ros2 topic list does not show the Create3 is publishing topics

Srimathi-tw, 2022-09-22:
**Please provide the following information:**
 - OS: Ubuntu 20.04
 - ROS Distro: Galactic
 - Built from source or installed: source 

 **Expected behaviour**
After following the setup instruction [https://turtlebot.github.io/turtlebot4-user-manual/overview/quick_start.html](https://turtlebot.github.io/turtlebot4-user-manual/overview/quick_start.html) on typing `ros2 topic list` it should show create 3 topics

 **Actual behaviour**
On typing `ros2 topic list` it runs in an infinite loop

**To Reproduce**
Followed the setup with turtlebot4 lite with [https://turtlebot.github.io/turtlebot4-user-manual/overview/quick_start.html](https://turtlebot.github.io/turtlebot4-user-manual/overview/quick_start.html)

Srimathi-tw, 2022-09-22:
The above mentioned resolution works with our personal hotspot. But we want to try using corporate hotspot, so we tried following this link [https://iroboteducation.github.io/create3_docs/setup/xml-config/](https://iroboteducation.github.io/create3_docs/setup/xml-config/)

We used the below code 
```
<CycloneDDS>
  <Domain>
    <Id>any</Id>
    <General>
      <NetworkInterfaceAddress>auto</NetworkInterfaceAddress>
      <AllowMulticast>false</AllowMulticast>
      <EnableMulticastLoopback>true</EnableMulticastLoopback>
    </General>
    <Discovery>
      <ParticipantIndex>0</ParticipantIndex>
      <Peers>
        <Peer Address="${ROBOT_IP}:7410"/>
      </Peers>
    </Discovery>
  </Domain>
</CycloneDDS>

```
and changed `${ROBOT_IP}` to our custom Robot IP, but on doing `ros2 topic list` we only see 
```
/parameter_events
/rosout
```
. Could you please help us with steps for connecting to corporate wifi?

roni-kreinin, 2022-09-22:
Did you use the RPi4 IP or Create 3 IP for the `${ROBOT_IP}`?

Srimathi-tw, 2022-09-22:
We were using raspberry pi IP address

roni-kreinin, 2022-09-22:
I believe for the create 3 instructions you should be using the Create 3's WiFi IP.

Srimathi-tw, 2022-09-27:
@roni-kreinin tried create 3's ip still didn't work. Any pointers on how we can connect to corporate wifi?

roni-kreinin, 2022-09-27:
I would recommend making an issue at https://github.com/eclipse-cyclonedds/cyclonedds. They will be much more familiar with this.

Srimathi-tw, 2022-09-29:
Thanks @roni-kreinin will close this issue then.

closed 2022-09-29 by Srimathi-tw