turtlebot/turtlebot4 #54: Connecting Turtlebot 4 to the virtual machine(UTM, VirtualBox,etc)

Tanjim13, 2022-09-23:
Hi, 
I am trying to connect my turtlebots using a virtual machine(UTM, VirtualBox, ). When I try it without virtual box(such as a directly Linux-installed laptop), I get the topic list from Turtlebot 4.  But I do not get the list when I try with the virtual machine installed Linux. Can you let me know the reasons and how to set up for virtual machine?

Tanjim13, 2022-09-23:
I am attaching some screenshots to make it clear. 1st screenshot is of .bashrc file of my VirtualBox terminal. The second ss shows one virtualbox terminal and one rpi's terminal.
[image]
[image]

tonybaltovski, 2022-09-23:
Is the VM set to `bridge-mode` for the networking adaptor?

Tanjim13, 2022-09-23:
No, it is in NAT mode.
[image]

tonybaltovski, 2022-09-23:
Using NAT will put the VM on a separate network whereas using bridge mode will put it on the same network.

Tanjim13, 2022-09-23:
@tonybaltovski Thank you so much. It worked.

closed 2022-09-23 by Tanjim13