turtlebot/turtlebot4 #55: Reason why the color video stream is disabled

alexswerner, 2022-09-23:
Is there a reason why the color stream is disabled (launch file argument `useVideo` in oakd.launch.py). The preview resolution is pretty low. Likely the reason is computation time, but I wish that there would be a way to get at least single frame with the full resolution.

roni-kreinin, 2022-09-26:
The video stream publishes the full 1080p video while the preview is just a crop of that full image. The RPi struggles to publish the full 1080p image with its internal WiFi at a reasonable framerate. The intention of the preview image is to assist with teleop.

You can modify the oakd [launch file](https://github.com/turtlebot/turtlebot4_robot/blob/galactic/turtlebot4_bringup/launch/oakd.launch.py) to set the camera parameters to whatever you like. Or you can try some of the other nodes at https://github.com/luxonis/depthai-ros-examples

alexswerner, 2022-09-30:
Thanks for the reply. I understand that streaming the full image over Wi-Fi will not work. In our application we want to implement tag detection on the Raspberry Pi, and that can be done locally on the raspberry pi. For better accuracy we want a higher resolution and accept a lower framerate instead. UseVideo/UsePreview in the launch file can enable that usage. The full resolution gives fps between 1 and 2.

closed 2022-09-30 by alexswerner

alexswerner, 2022-11-17:
I now learned that the camera has an API to capture a still image at full sensor resolution. However this is not made available in ROS: https://github.com/luxonis/depthai-ros-examples/issues/61