turtlebot/turtlebot4 #56: Turtlebot 4 lite raspberry pi wifi network not showing up. Unable to ssh into it.

OverlordFlowers, 2022-09-26:
**Please provide the following information:**
 - OS: Ubuntu 18.04
 - ROS Distro: Dashing
 - Built from source or installed: Installed
 - Package version:

 **Expected behaviour**
I'm expecting to see the Wi-Fi network "Turtlebot4" to appear when the TurtleBot 4 lite powers on.

 **Actual behaviour**
I am not seeing the Wi-Fi network appear, and am not able to connect and SSH into the raspberry pi.

**To Reproduce**
Provide the steps to reproduce:
1.  Place TurtleBot 4 lite onto charging dock to turn on.
2.  Wait 30 seconds for a noise to chime.
3. Look in available internet connections on host computer for TurtleBot 4.

**Other notes**
Add anything else you thing is important.

roni-kreinin, 2022-09-26:
The RPi creates a 5GHz network. Can you make sure that your WiFi adapter supports that?

If not, you can try to connect to it using an ethernet cable. See https://turtlebot.github.io/turtlebot4-user-manual/troubleshooting/faq.html#access-the-rpi-over-ethernet

OverlordFlowers, 2022-09-27:
My WiFi adapter does support 5 GHz. I am using the TP-Link Nano AC600 USB WiFi adapter.

I decided to use the ethernet cable and I was able to SSH into it and configure the WiFi, but I do not know if it is connecting successfully or not. Does the raspberry pi only support connecting to a 5 GHz network?

Additionally, may I ask what is the difference between the Create 3 and the Raspberry Pi, and why we need to configure the WiFi for both of them?

OverlordFlowers, 2022-09-27:
Additionally, while I was configuring the Turtlebot 4 standard, when it rebooted the display screen was off and none of the LEDs were working. Is it because it did not connect to a network successfully?

roni-kreinin, 2022-09-27:
The RPi supports both 2.4 and 5GHz networks but it is recommended to use 5GHz to improve performance for things like streaming camera data. 

To check if the RPi has connected to WiFi, try to SSH in and run `ip a`. You should see an entry for `wlan0` with an IP address if it is connected like so:
```
4: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether e4:5f:01:89:bc:59 brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.187/24 brd 192.168.0.255 scope global dynamic noprefixroute wlan0
       valid_lft 6476sec preferred_lft 6476sec
    inet6 fe80::e65f:1ff:fe89:bc59/64 scope link 
       valid_lft forever preferred_lft forever
```
If the interface is down, you can manually edit the network configuration located at `/etc/netplan/50-cloud-init.yaml`. Make sure that your SSID and password are correct. If you make any changes there, you can apply them by calling `sudo netplan apply`.

> Also, while I was configuring the Turtlebot 4 standard, when it rebooted the display screen was off and none of the LEDs were working. Is it because it did not connect to a network successfully?

We are working on a solution to allow the TurtleBot 4 to only use the RPi WiFi but it requires some IP routing that we do not yet support. See #26 for details. You are able to run the robot without connecting the Create 3 currently, but the ROS2 topics will only appear on the RPi4, and not an external PC.

> Also, while I was configuring the Turtlebot 4 standard, when it rebooted the display screen was off and none of the LEDs were working. Is it because it did not connect to a network successfully?

By default CycloneDDS is configured to use both `usb0` and `wlan0`, and will fail if one of the interfaces is down. If you check the logs on the RPi with `sudo journalctl -u turtlebot4 -r` you will likely see a bunch of errors from CycloneDDS.

OverlordFlowers, 2022-09-27:
If it shows:

4: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state DOWN group default qlen 1000
    link/ether e4:5f:01:bb:8f:2b brd ff:ff:ff:ff:ff:ff

that means it did not connect successfully, right?

OverlordFlowers, 2022-09-27:
I've modified the 50-cloud-init.yaml file and rebooted. I am no longer able to ssh into the raspberry pi using the ethernet cable.

roni-kreinin, 2022-09-27:
> If it shows:
> 
> 4: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state DOWN group default qlen 1000 link/ether e4:5f:01:bb:8f:2b brd ff:ff:ff:ff:ff:ff
> 
> that means it did not connect successfully, right?

Yes, it did not connect.

> I've modified the 50-cloud-init.yaml file and rebooted. I am no longer able to ssh into the raspberry pi using the ethernet cable.

What did you change? The ethernet and usb0 should not be modified.

OverlordFlowers, 2022-09-27:
I uncommented the “# ap” line to see if it would reboot in ap mode and deleted my ssid and password fields

roni-kreinin, 2022-09-27:
The AP mode still requires a SSID and password so this is likely an invalid configuration. Netplan probably can't run properly which is why the ethernet stopped working. For future reference, you can call `sudo wifi.sh -a` to set the RPi back to AP mode.

If you have a micro HDMI cable you can connect that to the HDMI0 port of the RPi. You can then use a monitor and keyboard to recover. Your other option is to just reflash the SD card by following the instructions [here](https://turtlebot.github.io/turtlebot4-user-manual/overview/quick_start.html#install-latest-raspberry-pi-image). The RPi should be back in AP mode with a fresh install.

Once you have recovered the RPi I would recommend creating a WiFi hotspot from your phone and trying to connect the RPi4 to it. If that works then the issue may be with your WiFi network.

OverlordFlowers, 2022-09-27:
I have been using a hotspot with my phone. I will try to recover the 50-cloud-init-yaml file. What’s the proper format?

roni-kreinin, 2022-09-27:
See [wifi.sh](https://github.com/turtlebot/turtlebot4_setup/blob/galactic/scripts/wifi.sh#:~:text=echo%20%2De%20%22,true%5Cn%22). I would recommend running `sudo wifi.sh -a` first before modifying so that the formatting is correct. You can also try to comment out the `band: 5GHz` line when in AP mode so that the AP is 2.4GHz.

OverlordFlowers, 2022-10-18:
I was able to connect it to a router that had a 5 GHz network but I am still unable to connect it to a 2.4 GHz network. I am also trying to connect it to my campus Wi-Fi. Is there any encryption that I should be aware of on the Turtlebot?

closed 2023-06-20 by roni-kreinin