turtlebot/turtlebot4 #57: Want to use turtlebot4 audio

Tanjim13, 2022-09-27:
Hi, I am trying to figure out how to work on the audio note of turtlebot4. I want to play some audio(Hi, How are you?) using the turtlebot 4 speaker. Is it possible? How to do so? Thank you.

roni-kreinin, 2022-09-27:
You can control the Create 3 speaker using an [Audio Note](https://github.com/iRobotEducation/irobot_create_msgs/blob/galactic/msg/AudioNote.msg) message, or the [Audio Note Sequence](https://github.com/iRobotEducation/irobot_create_msgs/blob/galactic/action/AudioNoteSequence.action) action. I don't believe the Create 3 currently supports playing an actual audio file.

Tauhid-Tanjim13, 2022-10-27:
Yes, I can only make beep beep noise with different hz. But I cannot convert text to speech. I think I will need to plug in a separate speaker to add this feature.

closed 2022-11-17 by Tanjim13