turtlebot/turtlebot4 #59: Communication LED is off, not working, why? How to fix it?

Tanjim13, 2022-09-29:
I am working with 6 Turtlebot 4 under the same router. Frequently, the COMM LED turns off automatically. I want know the reason. I have separated them with different ROS_DOMAIN_ID  What is the proper and fastest way to turn on or fix the issue with COMM Led?

roni-kreinin, 2022-10-03:
The COMM led being off indicates that the connection between the RPI4 and Create3 is not up. It may be that the Create 3 or RPi lose WiFi connection at some point. I would recommend trying to use FastRTPS as it will maintain the USB-C connection even if WiFi drops. Take a look at the tutorial [here](https://turtlebot.github.io/turtlebot4-user-manual/tutorials/multiple_robots.html#setting-the-environment-variable) for instructions on how to easily switch RMW's.

Tanjim13, 2022-10-05:
@roni-kreinin , I have followed the instruction. I did it by creating and running ros_config.sh script in /usr/local/bin of RPI 4 of the turtlebot. But the problem is when I separate two turtlebots with different ROS_DOMAIN_ID and set RMW_IMPLEMENTATION to rmw_fastrtps_cpp using ros_config.sh, under the same wifi router, the IP address of both robots starts to blink. The last 3digits of IP addresses start to exchange among each other. For example, Turtlebot1 with domain_id 1 has IP address 192.168.0.106 and is switching to Turtlebot2's (domain_id 5) IP address 192.168.0.119 and vice versa. The remaining turtlebots(4 of them still using cyclone_dds) start making chiming sounds but not blinking. If I shut down one turtlebot that is using fastrtps, then the blinking/exchanging of other turtlebot stops. 

What should I do?

roni-kreinin, 2022-10-05:
I forgot to add to the instructions that you also need to update your `install.py` script. You can find the updated script [here](https://github.com/turtlebot/turtlebot4_setup/blob/roni-kreinin/domain_id/scripts/install.py). Download it and move it to `/usr/local/bin/`, then run the `ros_config.sh` script again.

Tanjim13, 2022-10-05:
@roni-kreinin , It worked for those two turtlebot4. But when I tried with another one, the led lights, with led display is gone. But I could ssh into it using previous IP address.Then I ssh into it using ethernet cable with 192.168.185.3. But after using sudo wifi.sh -a it is showing me turtlebot4 wifi. But still led light and display is not working. How to fix this?

roni-kreinin, 2022-10-05:
`sudo wifi.sh -a` puts the RPi back into AP mode. You need to connect it to the same WiFi network as the Create3.

Tauhid13, 2022-10-05:
The problem is that turtlebot is working but I can not see the led display or light. I know how to connect to Create3 under same wifi. But LED light or display is not working.

roni-kreinin, 2022-10-05:
You can check the logs of the turtlebot4 service with `sudo journalctl -u turtlebot4 -r`. You might need to restart the service with `sudo systemctl restart turtlebot4`.

Tanjim13, 2022-10-05:
I solve it using this command. sudo apt-get install -y ros-galactic-rmw-fastrtps-cpp
Cause it is not installed in our turtlebots. I think this the reason when I switch to rmw_fastrtps_cpp, my led lights and display turns off.
I might be wrong. But it is solved after installing this package.

Tanjim13, 2022-10-06:
@roni-kreinin , I am still facing problem. It is very slow when students work with the six turtlebot4  simultaneously. So, should I add more routers? Or can I solve it in another way? Sometimes, COMM led is off for 20-30 seconds. What more can I do to make the turtlebot response smoothly under the same router? Or how many bots should I place under a router that will not slow it down?

roni-kreinin, 2022-10-06:
@Tanjim13 I would like to figure out if this is a bandwidth or signal strength issue. Can you answer the following questions:

- Does the COMM led turn off while the robot is driving around or when it is standing still? 
- How strong is the WiFi signal for both 2.4 and 5GHz in the area that you are testing? You can check with a phone or laptop.
- Next time the COMMs light turns off, can you retrieve the logs from the Create 3 [webserver](https://turtlebot.github.io/turtlebot4-user-manual/overview/quick_start.html#access-the-webserver-v013-or-higher) and upload them here?
- Can you also call `ros2 topic list` from a remote PC when the light turns off? To check that the RPi4 topics still publish.
- Can you check the framerate of the camera stream with `ros2 topic hz /color/preview/image`? It should be publishing at 30hz.

Tanjim13, 2022-10-07:
@roni-kreinin,

-  Sometimes, It goes off while driving (both in fastrtps & cyclonedds mode). But fastrtps mode, it comes back after few seconds. Another problem is that, it is very slow when we run programs using virtual machines in fastrtps mode without ssh into RPI compare to cyclonedds. 
- For 5GHz:, RSSI: -45dBm, Noise: -93dBm, TxRate: 867Mbps...  2.4GHz: RSSI: -34dBm, Noise: -88~-93dBm, TxRate: 145Mbps
I will try to answer rest of the questions, when I will work with them again.

Tanjim13, 2022-10-17:
@roni-kreinin , I tried to retrieve logs from create3 webserver, but it did not work. I tried 192.168.0.103:8080, but it did not let me go to the site.
ros2 topic hz /color/preview/image it worked fine.. I will investigate it further.
fastrtps is very slow compare to cyclonedds.

Tauhid-Tanjim13, 2022-10-27:
Hi @roni-kreinin , I want a suggestion. If I add routers to the turtlebot4 will it create any difference? Also if I use fastrtps do I need to change anything in my local linux machines? like changing in my ~/.bashrc file? I found fastrtps is very slow compare to cyclonedds. When I work alone, it is okay to work. But if multiple users in a same room working with 4-7 turtlebots simultaneously, I find comm stops working on few of them. Even in fastrtps it turns off and on.

Tanjim13, 2022-11-05:
@roni-kreinin, I can access to Create3 webserver using the IP address of the iRobot-8XXXXXXXXXX device from the Router's connected devices. So, I checked that the version of Firmware version is G.2.2. 
Another turtlebot4 has come to the lab recently and I can access the web server using the IP address of RPI with 8080 port. The firmware version of this turtlebot4 is G.3.1.

So, should I update the firmware of the create3 base? Will it solve the problem?

I will try to send you the log files from the webserver next time when we will work with more robots simultaneously and if the COMM goes off.

Tanjim13, 2022-11-08:
[image]

The other two turtlebots are connected to another router within the room.
[image]
[image]
[image]

roni-kreinin, 2022-12-09:
@Tanjim13 Sorry for the delay.

I would recommend updating any robots you have to G.3.1 to have all of the latest Create 3 features and bug fixes. As for accessing the webserver with the 8080 port, that is only installed by default on the 0.1.3 image in the link you mentioned. You can manually add it to an older image if you want by copying [this](https://github.com/turtlebot/turtlebot4_setup/blob/galactic/conf/webserver.service) file to `/etc/systemd/system/` and then calling `sudo systemctl enable webserver.service`.

Regarding the comms issue, the easiest solution is likely to just use 2 routers and have 3 robots on each. There is an update coming that adds discovery server support to the Create 3, but until then there isn't a great solution.

Tanjim13, 2022-12-22:
@roni-kreinin Thank you so much.

I will look into it.

roni-kreinin, 2023-03-06:
Closing as stale. Feel free to reopen if the issue persists.

closed 2023-03-06 by roni-kreinin