turtlebot/turtlebot4 #60: Place the plate higher

peppo97, 2022-10-03:
Hi, I need to place the plate higher. 
How can I do this? 
Where can I buy this taller component?

[image]
[image]

Best regards,
Giuseppe De Simone, PhD

apopielas, 2022-10-06:
Hi! You can use any M4X0.7 Male-Female Standoffs. [Mcmaster Carr has some that would work](https://www.mcmaster.com/spacers/threaded-standoffs/system-of-measurement~metric/system-of-measurement~metric/thread-size~m4/material~aluminum/). Any other hardware supplier will also likely have them.

Just get enough to adjust the height to what you need.

peppo97, 2022-10-07:
Thank you very much for your reply

closed 2022-10-07 by peppo97