turtlebot/turtlebot4 #61: Hardware and software problems

MathisGodard, 2022-10-03:
Hello, 

A custommer has 2 robots and he has some problems on his robots : 

- One of the two simply does not boot (the screen does not display anything, the LEDs remain off and the lidar does not launch).
- The second launches and we can connect to the raspberry without problem but it is impossible to control it. It seems that there is a problem with the battery and the motors (see the two attachments).

[image]

[image]

Best regards, Mathis.

roni-kreinin, 2022-10-03:
For the robot that does not boot, does the ring light around the power button turn on? If not, you can try to reset the Create3 battery by removing the 5 screws on the bottom of the robot, removing the bottom plate, and then removing the battery for 15 minutes. Then, reinstall the battery and reattach the bottom plate.

For the second robot, the COMMS led is off which indicates that the base and RPi4 are not able to communicate. Has the Create 3 been connected to WiFi? If so, make sure that the 2.4 GHz and 5 GHz networks are bridged.

roni-kreinin, 2023-03-06:
Closing as stale.

closed 2023-03-06 by roni-kreinin