turtlebot/turtlebot4 #63: Turtlebot4 Factory reset without opening

Thoneley, 2022-10-06:
- OS: Ubuntu 20.04
 - ROS Distro: Galactic

Need to flash/reset SD card on raspberry PI without opening the turtlebot casing. Currently the offset screws from our supplier seem locked in/locktited, so we need to access the SD card without potentially breaking the PCB board covering it.

apopielas, 2022-10-06:
Hi! The loctite we use for those is Loctite 243 and should come apart with hand tools. Can you take a photo of the screw and nearby areas?

Thoneley, 2022-10-07:
I don't have access right now to send a picture along, however we couldn't get access to the standoffs no matter what we tried. It seems as if the hidden bolt doesn't have a head we can easily access (or at all) under the pcb cover, and manually trying to twist the standoffs off did not work.

apopielas, 2022-10-07:
Ok! The standoffs are Male-female so there isn't a bolt on the underside holding it in. It screws into an insert into the plastic. Here is how I recommend removing it.

1. Remove the top plate. This will ensure the standoffs are fully free to move and also prevent any damage to the top plate. 
2. Grab vice-grips or multigrip pliers. Put some electrical tape on the teeth to avoid scratching the powder coat finish on the post. 
3. Tightly grip the standoff as close to the PCBA as possible ensuring no components will be damaged when undoing the standoff. 
4. Unscrew the standoff. 

Once you get it off it would be very helpful to get a picture of how much loctite was on there and how it looks.

roni-kreinin, 2023-03-06:
Closing as stale.

closed 2023-03-06 by roni-kreinin