turtlebot/turtlebot4 #64: Battery discharging and charging

simon-hauser, 2022-10-11:
Hi,
I wanted to double check with you if the runtime and charging speed I have is within the expectations.

Today I started our TB standard version with 47% charge. It was on for around 25 min, during which I drove maybe 30-40 meters, so the robot was mostly idle. The Lidar was always turning, however I only drove it manually with the joystick. After this, the charge was at 16% and I needed to abort an experiment to charge it again.

The robot then was on the charging dock for almost 3.5 hours and the charge now says 96%.

Is this consistent with runtime and charging speed from your experience? To me it seems that discharging is too fast and charging too slow, but this is of course quite subjective ;).

Thanks for your help,
best,
simon

simon-hauser, 2022-10-13:
Hi,
Thanks for the quick answer.
I needed to wait a couple of seconds after taking the robot off the dock before the `/battery_status` was getting published. I drove maybe 10 m, stopped the robot for 10 seconds, then echoed the status:

`header:
  stamp:
    sec: 1650546432
    nanosec: 129763746
  frame_id: ''
voltage: 15.217000007629395
temperature: 25.850000381469727
current: -1.121999979019165
charge: 1.7480000257492065
capacity: 2.0490000247955322
design_capacity: 2.0490000247955322
percentage: 0.8500000238418579
power_supply_status: 0
power_supply_health: 0
power_supply_technology: 0
present: true
cell_voltage: []
cell_temperature: []
location: ''
serial_number: ''
---
header:
  stamp:
    sec: 1650546432
    nanosec: 129763746
  frame_id: ''
voltage: 15.217000007629395
temperature: 25.850000381469727
current: -1.121999979019165
charge: 1.7480000257492065
capacity: 2.0490000247955322
design_capacity: 2.0490000247955322
percentage: 0.8500000238418579
power_supply_status: 0
power_supply_health: 0
power_supply_technology: 0
present: true
cell_voltage: []
cell_temperature: []
location: ''
serial_number: ''
---
`
How does this look to you? The robot has no payload on top of the acrylic plate and no additional hardware is connected. Camera is connected and Lidar is turning.

simon-hauser, 2022-10-13:
When the robot is on the dock, this is the echo `/battery_status`:
```
header:
  stamp:
    sec: 1650546966
    nanosec: 94047288
  frame_id: ''
voltage: 15.753000259399414
temperature: 26.850000381469727
current: 0.27000001072883606
charge: 1.7730000019073486
capacity: 2.0490000247955322
design_capacity: 2.0490000247955322
percentage: 0.8700000047683716
power_supply_status: 0
power_supply_health: 0
power_supply_technology: 0
present: true
cell_voltage: []
cell_temperature: []
location: ''
serial_number: ''
---
```

I could add that when I received the robot, I could not switch it off: the LED ring would blink 3 times but no sound was played and the robot just stayed on. I could solve this issue by resetting the battery and since then booting and shutting down seems to work properly.

roni-kreinin, 2022-10-13:
The charging current does seem to be fairly low. Typically we want to see about 1A when charging. Can you try a couple of tests to narrow down the issue?

- On the RPi run `sudo systemctl stop turtlebot4`. This will stop the `turtlebot4` service which runs all of the ROS2 nodes. Once the service is stopped, echo `/battery_status` again while the robot is off the dock.
- Unplug the [power adapter board](https://iroboteducation.github.io/create3_docs/hw/adapter/) from the Create 3. This will power off the RPi, PCBA, and lidar/camera. Then echo `/battery_status` with the robot off the dock. You can also try charging with the adapter board disconnected which should give you faster charge time.

simon-hauser, 2022-10-19:
Sorry for the late reply, I was a couple of days out of office.
Here is the `/battery_status` after running `sudo systemctl stop turtlebot4` while the robot is off the dock. The robot has no payload added:
```
header:
  stamp:
    sec: 1650546335
    nanosec: 745660753
  frame_id: ''
voltage: 15.340999603271484
temperature: 23.850000381469727
current: -0.7680000066757202
charge: 1.7719999551773071
capacity: 2.0490000247955322
design_capacity: 2.0490000247955322
percentage: 0.8600000143051147
power_supply_status: 0
power_supply_health: 0
power_supply_technology: 0
present: true
cell_voltage: []
cell_temperature: []
location: ''
serial_number: ''
```
And this is the echo after unplugging the power adapter board, robot is off the dock:
```
header:
  stamp:
    sec: 1650546565
    nanosec: 805095732
  frame_id: ''
voltage: 15.404999732971191
temperature: 24.549999237060547
current: -0.24699999392032623
charge: 1.7230000495910645
capacity: 2.0490000247955322
design_capacity: 2.0490000247955322
percentage: 0.8399999737739563
power_supply_status: 0
power_supply_health: 0
power_supply_technology: 0
present: true
cell_voltage: []
cell_temperature: []
location: ''
serial_number: ''
```
And lastly the echo of the robot on the dock, adapter board unplugged:
```
header:
  stamp:
    sec: 1650546698
    nanosec: 67249081
  frame_id: ''
voltage: 15.843999862670898
temperature: 24.950000762939453
current: 1.0889999866485596
charge: 1.7269999980926514
capacity: 2.0490000247955322
design_capacity: 2.0490000247955322
percentage: 0.8399999737739563
power_supply_status: 0
power_supply_health: 0
power_supply_technology: 0
present: true
cell_voltage: []
cell_temperature: []
location: ''
serial_number: ''
```
The charging current was increasing over a couple of seconds after putting it on the dock but then stays at over 1A.

simon-hauser, 2022-10-25:
Are these numbers more in line with a usual charging/discharging behavior?
Accelerating the charging speed by unplugging the adapter board could be a possibility for us, so that should not be too much of an issue; thanks for the tip. But I'm still wondering about the suggested runtime of 2.5 - 4 hours that still seem much longer than what we are getting, even without additional payload. Running `sudo systemctl stop turtlebot4` reduces the battery current by around 30%, but at the cost of stopping ROS2. Is there anything else I could try?
Thanks for your help,
best,
simon

simon-hauser, 2022-11-09:
Hi,
Sorry for poking you again with this topic, but we couldn't yet come to a conclusion of the issue.
Charging is largely solved; it should probably always be possible to remove the adapter board to speed up charging time.
However, discharing is still an issue. Is a discharge current of over -1.1 A normal under no payload? My estimated runtime is 1h, maybe a little bit more, but far from the 2.5-4h range. Is there anything else we can look into?
Thanks for your help.
Best,
Simon

roni-kreinin, 2022-11-09:
@simon-hauser Sorry for the delay. I will do some testing on my side to see what the typical current looks like when ROS2 is not running. 

In the meantime, you can use the [RPLIDAR motor](https://github.com/turtlebot/turtlebot4_robot/blob/galactic/turtlebot4_bringup/config/turtlebot4.yaml#:~:text=%23%20EStop-,%23%20RPLIDAR%20Motor,-%23%20Menu%20Functions) function to stop/start the RPLIDAR. The motor on the RPLIDAR draws about 0.4A itself, so if you are not actively using it you can stop it to save power.

slowrunner, 2022-11-26:
For reference my TB4lite draws 13W with the motor running.

My undocked sessions go from full charge to 20% red ring in roughly 1 hour 50 minutes, 
and from 96% to 24% in roughly 1 hour 45 minutes.
Charging from 24% to 96% takes roughly 3 hours. 

```
2022-11-25 17:36|wali_node.py| ** WaLI Undocking: success at battery 96% **
2022-11-25 19:21|wali_node.py| ** WaLI Docking: success at battery 24% **
2022-11-25 22:19|wali_node.py| ** WaLI Undocking: success at battery 96% **

*** TOTAL WaLI Turtlebot 4 Lite LIFE STATISTICS ***
Total Life:  135.25 hrs (since Nov 17, 2022)
Sessions (boot):  16
Average Session:  8.4 hrs
Successful Dockings Logged:  22
```

closed 2023-04-14 by roni-kreinin