turtlebot/turtlebot4 #65: The power light and LCD display are off

axdwq, 2022-10-12:
- OS: Ubuntu 20.04
 - ROS Distro:  Galactic
 - installed: already installed

 **Expected behaviour**
Following the quick start guide after I changed the wifi id and password, I rebooted. It should rebooted and connect again but it isn't.

 **Actual behavior**
Raspberry pi 4 do not start and the power light and LCD display are off.

**To Reproduce**
I am new in this so, I tried to restart removing the turtlebot4 from the dock and no longer worked again.

**Other notes**
I am working around Ethernet connection but I don't know the password. It must be "ubuntu" and it isn't. Can you let me know what is the password for Raspberry pi Ethernet connection?

roni-kreinin, 2022-10-12:
The password is `turtlebot4`

axdwq, 2022-10-13:
I already test the password turtlebot4 and do not works.
What can I do?

roni-kreinin, 2022-10-13:
Did you change the username or password when you were setting up wifi? Because you would have had to use `ubuntu` as the username and `turtlebot4` as the password to SSH into the robot in the first place. The username and password should be the same regardless of if you are SSH'ing over wifi or ethernet.

axdwq, 2022-10-13:
I mean angle brackets around the user name and password, when We want to change those.

roni-kreinin, 2022-10-13:
To clarify, are you referring to changing the username and password of the ubuntu user? Or the SSID/password of the WiFi network you were connecting to with `sudo wifi.sh`?

axdwq, 2022-10-14:
I was trying to change WIFI_SSID and WIFI_PASSWORD on wifi.sh file.

roni-kreinin, 2022-10-14:
In that case, the SSID and password should be that of your WiFi network, and not `turtlebot4`. You should still be able to access the Pi over SSH by connecting with ethernet and running `ssh ubuntu@192.168.185.3` as instructed [here](https://turtlebot.github.io/turtlebot4-user-manual/troubleshooting/faq.html#access-the-rpi-over-ethernet). When prompted for a password, enter `turtlebot4`. Then you can run the WiFi setup again and use your WiFi credentials. Also you should use single quotation marks like 'this' around the SSID/password. 

For example if your WiFi network has an SSID of MyNetwork and password MyPassword while in the USA, you would run:
`sudo wifi.sh -s 'MyNetwork' -p 'MyPassword' -r US && sudo reboot`

axdwq, 2022-10-14:
Then I did every thing well. I wrote: sudo wifi.sh -s 'Turtlebot4' -p 'Turtlebot4' -r MX && sudo reboot. After that,  the raspberry pi do not rebooted well.
Now, I can't see the Turtlebot4 name in my WiFi Network and the LCD display is still off.

roni-kreinin, 2022-10-14:
Your WiFi network is called Turtlebot4 with password Turtlebot4?

axdwq, 2022-10-18:
Yes, it  is

brandon-todd, 2023-01-20:
you need to set the AP to 1 for it to show the network. Try the wifi.sh -s Turtlebot4 -p Turtlebot4 -r US -a 1 && sudo reboot cmd

axdwq, 2023-02-24:
Thank you for your help. I already fixed the issue. 
What I did, was to reflash the RPi4 SD Card. After that, I got access to RPi4 and I changed the middleware from cyclonedds to fastrtps. Then, I restarted the Turtlebot 4 and the LCD display and the five leds turned on correctly. Now, the turtlebot 4 is working right.

closed 2023-03-22 by roni-kreinin