turtlebot/turtlebot4 #66: turtlebot cannot drive backwards with joystick

simon-hauser, 2022-10-13:
Hi,
I can't really make the robot drive backwards when controlling the turtlebot with the provided joystick, neither in or out of the safety mode. It would drive back maybe 10 cm, then stutter, then drive another 2 cm and then stop completely. After that, the robot refuses to drive backwards entirely. I need to drive forward (or turn) before another backwards drive can be attempted, but the behavior is always the same.
I have not tried driving backwards with other methods, but I wanted to clarify if this is the correct behavior?

roni-kreinin, 2022-10-13:
How are you disabling the safety? I typically use the [webserver](https://iroboteducation.github.io/create3_docs/webserver/application/) to set `safety_override` to `full`.

simon-hauser, 2022-10-17:
Ah, that was unclear from my side. I meant using L1 or R1 on the controller to drive the robot. From my understanding, L1 is driving with safetly enabled, and R1 is driving without safety.
I have not tried to disable the safety via webserver. But since the driving back problem also exists with safety, I would guess that this is not the main issue?
I might add that after the robot stopps when trying to drive backwards, the LED ring lights up yellow (I think, I'd have to check the exact color again). A similar LED color change happens when the robot is kidnapped, but it seems a bit unlikely that the backwards driving would trigger a kidnap error (which probably uses the wheel suspension for its trigger?).

roni-kreinin, 2022-10-19:
L1 simply limits driving and turning velocities, while R1 allows for faster speeds. Out of the box the Create 3 will have safety enabled and this needs to be disabled via the webserver to allow backwards driving. 

The yellow lightring indicates that the wheels have been stopped. This occurs when you reach the backup limit, kidnap the robot, set the E-Stop, and some other situations.

simon-hauser, 2022-10-19:
Ok, thanks. I will try out disablinng the safety and report back if this worked.
Out of curiosity, into which error does the robot run when driving backwards with safety enabled? Is it maybe that the robot doesn't have cliff sensors for safe backwards driving and can only retract a maximal distance (is this what you mean by backup limit)?

roni-kreinin, 2022-10-19:
Yes, the cliff sensors are located on the front half of the robot and prevent it from falling down. If you disable safety then the cliff reflex will be disabled and the robot can fall down stairs or ledges if you are not careful. If you want to keep safety enabled, you can simply rotate the robot 180 degrees on the spot and drive forwards instead of reversing. Also, the undock action will still allow the robot to reverse from the dock even with safety enabled.

simon-hauser, 2022-10-19:
Disabling the safety via the webserver worked. Thanks for your help!

closed 2022-10-19 by simon-hauser