turtlebot/turtlebot4 #67: Facing issues with turtlebot4_ignition_bringup

Tanjim13, 2022-10-17:
Whenever I run this command: My turtlebot4 starts to blink **comm, wifi and batt leds.**  
And ip address switches between UNKNOWN and 192.168.0.103 (turtlebot's ip) very rapidly.
ros2 launch turtlebot4_ignition_bringup ignition.launch.py
What is the reason behind it?
How to solve it?

roni-kreinin, 2022-10-19:
You need to use a different ROS_DOMAIN_ID if you are running the simulation and a real robot concurrently. Otherwise the ros2 topics of the simulation will interfere with the robot.

Tanjim13, 2022-10-22:
Okay, I will try and let you know. Thank you.

Tanjim13, 2022-11-17:
@roni-kreinin, I have not checked it yet. If I failed to resolve it, I will open it again. Now I am closing this issue. Hope it will work.

closed 2022-11-17 by Tanjim13