turtlebot/turtlebot4 #68: Battery pack issue

alexswerner, 2022-10-17:
From our fleet of 15 Turtlebot4 one unit seems to have an issue with the battery. The behavior is that the robot is charging the battery to full, but then stops charging for some reason. The battery is discharged until the Raspberry Pi turns off and the red ring status light appears. This happens while the robot is continuously docked. Re-docking the robot restarts the cycle. I swapped the battery of the problematic robot to another one, and now this robot shows the same behavior. So, it seems that the battery pack has an issue.
This happens to be the battery of the first unit we received for testing. The 14 units of the next shipment work fine in this regard.
Let me know how to proceed.

roni-kreinin, 2022-10-19:
Can you send the logs from the Create 3 webserver after this occurs?

alexswerner, 2022-10-25:
Sure, here is the log (after reboot) 
[battery_issue.txt](https://github.com/turtlebot/turtlebot4/files/9860504/battery_issue.txt)

roni-kreinin, 2022-11-15:
@alexswerner Have you tried resetting the battery by taking it out of the robot for at least 15 minutes? If not, could you try that?

In the meanwhile I will ask iRobot if they have seen this issue before.

alexswerner, 2022-11-17:
I did that, but it did not help. The issue was very persistent for at least a month. It was also reproducible when moving the battery pack to a new robot. However, the problem has not occured since some weeks. We'll wait and see.

closed 2022-11-17 by alexswerner

roni-kreinin, 2022-11-17:
@alexswerner If the issue occurs again please contact our support team at [support@clearpathrobotics.com](mailto:support@clearpathrobotics.com) and we can RMA the battery.