turtlebot/turtlebot4 #69: how can i get data from turtlebot4 clearpath to remote pc

HSmoon1031, 2022-10-19:
**Please provide the following information:**
 - OS: turtlebot4 Clearpath Ubuntu 20.04 & remote pc ubuntu 20.04
 - ROS Distro: turtlebot4 - galactic & remote pc - galactic
 
 **Expected behaviour**
      i expected 'ros2 topic echo /odom' on remote pc 
      i thought it can receive data. 
     but there is no /odom 

    i set up on turtlebot4 : 
              bashrc -> 'export RMW_IMPLEMENTATION=rmw_fastrtps_cpp'  
              
   remote pc only conneting wifi, and ros2 

  and turtlebot can move by my wireless controller ps4 
   
  and when run to rviz on remote pc , there is no global status. and robotmodel 

  and i tried to pub like this 'ros2 topic pub --rate 1 /cmd_vel geometry_msgs/msg/Twist "{linear: {x: 0.5, y: 0.0, z: 0.0}, angular: {x: 0.0, y: 0.0, z: 1.0}}"' on remote PC then no action 
 
 but on turtlebot4 did 'ros2 topic pub --rate 1 /cmd_vel geometry_msgs/msg/Twist "{linear: {x: 0.5, y: 0.0, z: 0.0}, angular: {x: 0.0, y: 0.0, z: 1.0}}"'  is working 

 and i use our place iptime, turtlebot connected 5Ghz, my computer Pc using wirelessLan receiver connected 5Ghz, create3? i don't know  

so how can i get the data from turtlebot4 to remote pc . what is problem?? plz help me... i spent time too many T.T

 **Actual behaviour**
 ssh-> into turtlebot4 clearpth : 
     $ros2 topic list 
           result:
                /battery_state
                /cmd_audio
                /cmd_lightring
                /cmd_vel
                /color/preview/camera_info
                /color/preview/image
                /diagnostics
                /diagnostics_agg
                /diagnostics_toplevel_state
                /dock
                /hazard_detection
                /hmi/buttons
                /hmi/display
                /hmi/display/message
                /hmi/led
                /imu
                /interface_buttons
                /ip
                /ir_intensity
                /ir_opcode
                /joint_states
                /joy
                /joy/set_feedback
                /kidnap_status
                /mouse
                /odom
                /parameter_events
                /robot_description
                /robot_state/transition_event
                /rosout
                /scan
                /slip_status
                /static_transform/transition_event
                /stereo/camera_info
                /stereo/depth
                /stop_status
                /system_monitor/transition_event
                /tf
                /tf_static
                /wheel_status
                /wheel_ticks
                /wheel_vels

 remote pc :
      $ros2 topic list 
       result : 
            /battery_state
            /cmd_vel
            /color/preview/camera_info
            /color/preview/image
            /diagnostics
            /diagnostics_agg
            /diagnostics_toplevel_state
            /dock
            /hazard_detection
            /hmi/buttons
            /hmi/display
            /hmi/display/message
            /hmi/led
            /imu
            /interface_buttons
            /ip
            /joint_states
            /joy
            /joy/set_feedback
            /mouse
            /parameter_events
            /robot_description
            /rosout
            /scan
            /stereo/camera_info
            /stereo/depth
            /tf
            /tf_static
            /wheel_status
    

**To Reproduce**
Provide the steps to reproduce:
1. run something
2. launch something else
3. see the error

**Other notes**
Add anything else you thing is important.

HSmoon1031, 2022-10-24:
hello 
maybe remote pc cannot communication with Create3 Data , just RPI4 data. 
on remote pc , only can echo /cmd_vel, can not echo /odom

HSmoon1031, 2022-10-26:
anybody there~~~~~~~~~~~~~? plz give me hint~~~~~~~~~~~~~```

roni-kreinin, 2022-10-31:
The Create3 needs to be connected to a 2.4GHz network that is bridged with your 5GHz network. Otherwise data will only be sent over USB to RPi4, and not to your remote PC.

closed 2023-04-14 by roni-kreinin