turtlebot/turtlebot4 #70: Displaying wrong WIFI IP address

teufelweich, 2022-10-25:
**Please provide the following information:**
 - Built from source or installed: freshly installed image v0.1.3
 - Package version: v0.1.3

 **Expected behaviour**
When the WIFI connection fails (e.g. the interface has no ip adress) it should not display the default `10.42.0.1` on the Turtlebots display. But maybe some kind of error or its ethernet IP address (like `eth: 192.168.185.3`).

 **Actual behaviour**
default `10.42.0.1` is displayed

**To Reproduce**
Provide the steps to reproduce:
1. install image
2. configure invalid WIFI credentials
3. reboot

```bash
ubuntu@ubuntu:~$ ip link show wlan0 
4: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state DOWN mode DORMANT group default qlen 1000
    link/ether e4:5f:01:7d:56:4f brd ff:ff:ff:ff:ff:ff
```

roni-kreinin, 2022-10-31:
The display should typically display "UNKNOWN" as the IP address in a case like this. I think the issue here is that CycloneDDS is configured by default to use the `wlan0` and `usb0` interfaces. If either one of these fails then CycloneDDS throws errors and ROS2 nodes will crash. If you use FastDDS then this issue should not occur, and "UNKNOWN" should be displayed correctly.

I'm assuming you are rebooting just the RPi4, in which case the turtlebot4 node will not start up properly and will not update the display. The display will continue to show the last image that was sent to it, which includes the `10.42.0.1` IP address. Fully power cycling the robot will result in the display being completely off until WiFi is working.

CycleMark, 2022-11-21:
Not sure if it's of any use - just started with my TB4 - problem I had with the Pi is that it wouldn't connect to my WiFi until I ran the command to bring the interface up - 

nmcli c up <SavedWiFiConn>

Mark

roni-kreinin, 2022-11-21:
@CycleMark Did you connect to WiFi using the `wifi.sh` script?

CycleMark, 2022-11-21:
Yep - I was following the instructions as per - https://www.youtube.com/watch?v=QN01AX