turtlebot/turtlebot4 #72: Led light is black cannot ssh into Turtlebot4

Tanjim13, 2022-10-28:
I have tried Ethernet using ssh ubuntu@192.168.185.3 to get into robot, but failed. I can see ros topics of the turtlebot from my terminal using ROS_DOMAIN_ID (but /ip is missing). I enter into create3 log file to see if there is any wrong. Posting that below(couldn't understand). I am not so sure if anyone has worked on it. How can I solve this issue? Thanks in advance.

Create3 log:
Oct 28 07:00:31 iRobot-8713D68C91464F59AE89344BCD88E3CB daemon.err network[1474]: [ERROR] (src/wifi_control.c:207 errno:Address family not supported by protocol) sudo network_device_interfaces_control.sh --interface=softap --action=del returned status 1
Oct 28 07:00:44 iRobot-8713D68C91464F59AE89344BCD88E3CB daemon.err network[1474]: [ERROR] (src/aos_kvs_helper.c:51 errno:No such file or directory) Failed to stat file robot.disc_reconnect_sleep_sec
Oct 28 07:00:44 iRobot-8713D68C91464F59AE89344BCD88E3CB daemon.err network[1474]: [ERROR] (src/kvs_json.c:21 errno:No such file or directory) Failed to retrieve key "robot.disc_reconnect_sleep_sec"
Oct 28 07:00:44 iRobot-8713D68C91464F59AE89344BCD88E3CB daemon.err network[1474]: [ERROR] (src/aos_kvs_helper.c:51 errno:No such file or directory) Failed to stat file network.watchdog_interval_min
Oct 28 07:00:44 iRobot-8713D68C91464F59AE89344BCD88E3CB daemon.err network[1474]: [ERROR] (src/kvs_json.c:21 errno:No such file or directory) Failed to retrieve key "network.watchdog_interval_min"
Oct 28 07:00:44 iRobot-8713D68C91464F59AE89344BCD88E3CB daemon.err network[1474]: [ERROR] (src/aos_misc.c:86 errno:No such file or directory)  MODULE : network_manager CMD (sudo network_dhcp_arp_control.sh disable) failed with status = 1, ret = 256
Oct 28 07:00:48 iRobot-8713D68C91464F59AE89344BCD88E3CB daemon.err network[1474]: [ERROR] (src/mqtt_client.c:695) firr_transfer failed
Oct 28 07:00:48 iRobot-8713D68C91464F59AE89344BCD88E3CB daemon.err network[1474]: [ERROR] (src/main.c:169) Error publishing sta:0 FIRR request.
Oct 28 07:00:49 iRobot-8713D68C91464F59AE89344BCD88E3CB daemon.err network[1474]: [ERROR] (src/wifi_control.c:207 errno:No child process) sudo network_device_interfaces_control.sh --interface=softap --action=del returned status 1
Oct 28 07:09:32 iRobot-8713D68C91464F59AE89344BCD88E3CB daemon.err network[1474]: [ERROR] (src/mqtt_client.c:695) firr_transfer failed
Oct 28 07:09:32 iRobot-8713D68C91464F59AE89344BCD88E3CB daemon.err network[1474]: [ERROR] (src/discovery.c:407) falied to get cap from tsprime
Oct 28 07:11:31 iRobot-8713D68C91464F59AE89344BCD88E3CB daemon.err network[1474]: [ERROR] (src/mqtt_client.c:319 errno:No such file or directory) Failed to delete /tmp/cleantrack_connected
Oct 28 07:11:35 iRobot-8713D68C91464F59AE89344BCD88E3CB daemon.err network[1474]: [ERROR] (src/mqtt_client.c:695) firr_transfer failed
Oct 28 07:11:35 iRobot-8713D68C91464F59AE89344BCD88E3CB daemon.err network[1474]: [ERROR] (src/provisioning_sm.c:637) Error publishing {"event":"disconnect_local"} firr request.

roni-kreinin, 2022-10-31:
Are you able to check the IP address of the RPi using your routers webserver? If not, can you try using `nmap` as instructed [here](https://turtlebot.github.io/turtlebot4-user-manual/overview/quick_start.html#find-the-raspberry-pi-ip-on-your-network)?

Tanjim13, 2022-11-05:
Hi, I will get that Turtlebot4 a couple of weeks later, so I will not be able to work on that for now. I am closing the issue. If I get it back and cannot work on it, I will open the issue again. Thank you.

closed 2022-11-05 by Tanjim13