turtlebot/turtlebot4 #73: Laptop - Raspberry Pi connectivity problems, black LED on Raspberry Pi.

lenaNzrva, 2022-10-28:
We use:
- Ubuntu 20.04
- Ros2 Galactic

We followed the "Quick Start" manual and everything was going smoothly until we executed the following command with our WIFI_SSID and WIFI_PASSWORD parameters:

`sudo wifi.sh -s '<WIFI_SSID>' -p '<WIFI_PASSWORD>' -r <REGULATORY_DOMAIN> && sudo reboot`

After this command, the robot rebooted but Raspberry Pi could not connect to our Wifi network. We tried to reach it by ethernet according to the manual. Specifically, we connected through SSH to Raspberry Pi using following:

`ssh ubuntu@192.168.185.3`

After connection, terminal was responding very slowly to our commands but we managed to restore AP mode of Raspberry Pi (Turtlebot4 network). We used this:

`sudo wifi.sh -ap 1 && sudo reboot`

 After that, LED became black, lamp on Raspberry Pi bord lighted with red, and we could not connect anymore to the boad by any means (neither wifi nor ethernet nor usb). As a first attmept, we tried to reboot both robot and laptop to try reconnect them (did not work for us). Can you give us some advices in this situation?

Thank you

roni-kreinin, 2022-10-31:
The command for restoring AP mode is `sudo wifi.sh -a`. By using `sudo wifi.sh -ap 1` you actually set the password to '1' and SSID to an empty string. The `wifi.sh` script does not currently error check so I believe this would cause an invalid YAML configuration for netplan. As a result, all of the network interfaces (`wlan0`, `eth0`, `usb0`) will likely be in an error state. 

Your two options for recovery are to either connect the RPi4 to a monitor via mini-HDMI and log in with a keyboard, or to reflash the SD card with a fresh image as instructed [here](https://turtlebot.github.io/turtlebot4-user-manual/overview/quick_start.html#install-latest-raspberry-pi-image).

Alaric102, 2022-11-01:
Hi, I have pretty same issue. I have reflashed the SD card using this instruction [here](https://turtlebot.github.io/turtlebot4-user-manual/overview/quick_start.html#install-latest-raspberry-pi-image). But RPi4 doesn't react when I turn it on. The output after 
`bash sd_flash.sh /path/to/downloaded/image`
`mmcblk0`
is: 
[image]

When I turn on the RPi4 the red diode is flashing stable, the green one is flashing for 5-10 seconds (I have read that it must flash ~20 sec) and after blinking. My screen light is also blinking. 

What should I do? Also, why is your Raspberry Pi image not in the IMG format?

closed 2023-04-14 by roni-kreinin