turtlebot/turtlebot4 #75: Multiple Turtlebot 4 robots on the same network (suggestions towards purchasing the best router)

hamidosooli, 2022-11-01:
We recently bought five Turtlebot 4 robots for which we are planning to buy a router. Since both iRobot base (create 3) and the top part need to be connected to the same network, and the base only can connect to 2.4 GHz wifi network. to connect five Turtlebots to the same network (with top and bottom sections connections) we need your suggestions towards purchasing the best router.

roni-kreinin, 2022-11-03:
I have had a good experience with this router https://www.tp-link.com/ca/home-networking/wifi-router/archer-c1200/

hamidosooli, 2022-11-03:
> I have had a good experience with this router https://www.tp-link.com/ca/home-networking/wifi-router/archer-c1200/

With how many robots? For one robot a simple model like [1] works fine (we tested with this one in our lab). However according to [2] for multiple robots it is another story.

[1] https://www.amazon.com/dp/B01N5RCZQH/ref=pe_386300_442618370_TE_dp_i1
[2] https://turtlebot.github.io/turtlebot4-user-manual/tutorials/multiple_robots.html

roni-kreinin, 2022-11-03:
I have had at least 3 running at once without issue. I'm not sure what you are referring to in the multiple robots tutorial.

hamidosooli, 2022-11-03:
Thank you very much for your comments. In the tutorials it is said that "If we connect two default TurtleBot 4's to the same network, the topics from both robots will communicate with each other and cause unwanted behaviours." Then I was worried if we connect 5 robots to the same network it may cause complications.

roni-kreinin, 2022-11-07:
You will have to configure each robot to have a unique domain ID, as instructed in the multiple robots tutorial. I would just recommend setting up one robot at a time as they all default to a domain ID of 0.

hamidosooli, 2022-11-16:
I just accomplished communicating 5 Turtlebots through the same network (through a simple router model [1]). As the number of robots increases the possibility for them to disconnect from the network goes up. I mean keeping all the status LEDs on for all the robots (especially COMM and BATT) becomes trickier.
[1] https://www.amazon.com/dp/B01N5RCZQH/ref=pe_386300_442618370_TE_dp_i1

closed 2022-11-29 by hamidosooli