turtlebot/turtlebot4 #79: Create3 not publishing topics

vatsuak, 2022-11-15:
Hi, I am using a Nvidia jetson running ubuntu 20.04 to connect to the turtlebot. I have followed the instructions in the quick start guide. 
I am using a wifi hotspot (on wlan0) from the Jetson itself. It is set to 2.4ghz so that both the pi and the base can connect to the same hotspot. I am using the default cyclonedds protocol over wlan0
However, I am having trouble connecting to the create3 and cannot receive any of the topics from it. Is there something I am missing?

roni-kreinin, 2022-11-16:
Can you access the Create 3 [webserver](https://turtlebot.github.io/turtlebot4-user-manual/overview/quick_start.html#create-3-wifi-setup) and check that the Create 3 is configured for cyclonedds with the same ROS_DOMAIN_ID as your RPi?

vatsuak, 2022-11-18:
Yes, I think there was an issue with the create base not being able to connect to the wifi. Using a correct connection and keeping all the domain IDs as default I was able to get the desired data. Also, it seems that even if we don't connect the create3 to the wifi, the topics are visible on `ros2 topic list` however we can't echo them. Maybe the USB connection between the base and the pi is the reason for this.  

I had a few more questions for the connection issue:
1. How can we install packages like Bluetooth and others without needing to connect the pi to the internet
2. Can we set the pi to AP mode without sshing (sometimes if we enter some incorrect network property for the wifi, it just keeps on searching and we cannot ssh into it any more)
3. For getting the sensor readings, off the OAK-D and the lidar, can we not simply get read the topic to get the measurements? I can see the topics like depth and preview but cant visualize them normally in rviz2

roni-kreinin, 2022-11-18:
1. Bluetooth packages should already be installed. Try running `sudo bluetoothctl` to pair a controller or bluetooth device. To install other packages without internet connection you will need to download the .deb files to your laptop, then use `scp` to copy them to the RPi. Then you can install them with `dpkg`.
2. There is no way to reset the Pi back to AP mode without ssh'ing in or hooking it up to a monitor, other than by flashing a fresh turtlebot4 image onto the SD card. To recover from an incorrect WiFi configuration, try SSH'ing using an Ethernet cable and following the instructions [here](https://turtlebot.github.io/turtlebot4-user-manual/troubleshooting/faq.html#access-the-rpi-over-ethernet).
3. You can get the raw data of the lidar and OAK-D by using `ros2 topic echo`. For the lidar, call `ros2 topic echo /scan`.

vatsuak, 2022-11-19:
I just tried calling the `/color/preview/image` to view the image in RViz however I am getting the following message on the terminal,

`[INFO] [1668814362.569177576] [rviz]: Message Filter dropping message: frame 'oakd_pro_rgb_camera_optical_frame' at time 1650546203.787 for reason 'discarding message because the queue is full'`

I can, however, see the topic by calling `ros2 topic echo /color/preview/image`.

roni-kreinin, 2022-11-21:
@vatsuak Rviz can be a bit finnicky with camera images. I would recommend running `ros2 run rqt_image_view rqt_image_view`. 

Regarding bluetooth, you can run `sudo bluetooth.sh` which will just install some packages required for bluetooth. Then just reboot the RPi and you should be able to run `bluetoothctl`.

closed 2023-01-10 by vatsuak