turtlebot/turtlebot4 #85: Need a "SAFE" way to shutdown the TurtleBot4 Lite

slowrunner, 2022-11-26:
When I use the Create3 /power_off action goal, it jerks power out from under the Raspberry Pi4, corrupting files on the way down.

I have submitted an idea for the Create3 to include a delay on the /power_off goal so that the Raspberry Pi4 would have time to complete a proper shutdown.

THIS IS A VERY BAD PROBLEM.

(I tried shutting down the RPi4, then holding the Create3 power button till all goes dark, but sometimes this restarts everything.)

roni-kreinin, 2022-11-28:
We could definitely add a simple shutdown script that will send the /power_off goal and shut down the RPi before the robot powers off, but yes that would require iRobot to add a delay option to the service.

slowrunner, 2023-02-06:
iRobot has the corresponding enhancement issue 
https://github.com/iRobotEducation/create3_docs/issues/280