turtlebot/turtlebot4 #90: Nav2 not working correctly

ebear11, 2022-12-09:
**Please provide the following information:**
 - OS: (e.g. Ubuntu 20.04) Ubuntu 20.04
 - ROS Distro: (e.g. Galactic) galactic

 When I run the Nav2 command found in the documentation ("ros2 launch turtlebot4_navigation nav_bringup.launch.py slam:=off localization:=true map:=office.yaml")  I am left with a global error "no tf data found." I also notice the comm light on the turtlebot4 turns off. 
The create 3 logs say "ddsi_udp_conn_write to udp/192.168.1.126:46503 failed with retcode -3." (It repeats that with many different port numbers) I was able to save the map and actually map the environment fine but whenever I run the command found in the guide the comm light always turns off and that same error appears in the create 3 logs. I looked at issue #59 but I could not try switching the middleware as I could not find any instructions on how to configure the PC to use Fast RTPS. I am just a high school student trying to figure out the workings of how ROS2 works.

roni-kreinin, 2022-12-09:
To us FastRTPS on your PC you just need to `export RMW_IMPLEMENTATION=rmw_fastrtps_cpp`. If you don't have fastrtps installed you can install it with `sudo apt install ros-galactic-rmw-fastrtps-cpp`.

ebear11, 2022-12-14:
Thank you for the reply. I was able to successfully switch the middleware and I tested that it worked with manually driving it around, but I am still unable to navigate. The error that seems to be popping up in rviz is frame [map] not found. To be more specific, the terminal says,"timed out waiting for transform between base_link and odom to become available" The other part of the error mentions that the frame odom being passed to canTransform and that the target_frame does not exist. The light still turns off but with a new symptom. The main light turns solid red for a few seconds then back to normal but I can not drive the robot around after that. sometimes when it happens I can't even turn the robot off. The power button does not do anything no matter what I try to it.  I am almost starting to wonder whether or it is a hardware fault but I am going to assume it is something I am doing wrong.

JasmineRosselliSUPSI, 2023-01-19:
I recently had a similar problem after updating the create3 (but only a few weeks leater, so it my not be the cause) to the newer version (G.3.1). I have the versions ubuntu 20.04 Focal and ROS2 Galactic installed on the Rpi of my turtlebot4.

The similarities are:
- solid red light and turtlebot4 unable to react to any stimulus
- navigation issues
- the turtlebot4 CAN be driven using '`ros2 run teleop_twist_keyboard teleop_twist_keyboard`'
- the turtlebot4 CAN create maps (`ros2 launch turtlebot4_navigation slam_sync.launch.py`)
- the turtlebot4 CANNOT be driven using 'r`os2 launch turtlebot4_navigation nav_bringup.launch.py slam:=off localization:=true map:=<map_name>.yaml`' 

One more problem I have:
- /docking and /undocking don't work from script or through command line (action send_goals) -> BUT the robot is actually able to perform /dock and /undock using create3 physical buttons.

If run from script or terminal, it returns message:
'Goal_rejected' 

If turtlebot4 is not on docker station, always respond with '`Goal_rejected`' but sometimes perform dock anyway.

---
Here some more details
--- 

At first I did the update of create3 to solve nav2 a problem (often amcl_pose parameter wasn't publishing and the only way to fix that was to reboot the robot multiple times).

BUT! 

I now face too with navigation issues, with solid red light and turtlebot4 unable to receive or send commands, even from physical buttons. To momentarily fix the issue, the only thing that worked was to take off the battery and reboot the system.

I also started to notice problems at action and service level, which seem to be cumulative. The more action and service servers are compromised, the more others follow.

From create3 log i receive both retcode -3 and retcode -1 errors. I already tried to install older version of create3 and back to new, but in neither case worked.

If I test turtlebot4 behavior running '`ros2 run turtlebot4_tests ros_tests`',
I get stuck at tests 3 and 4.
The messages I get are basically the same (here some examples):

- [WARN] [1674139895.970490189] [turtlebot4_ros_tests]: Ignoring unexpected goal response. There may be more than one action server for the action '/drive_distance' 

- [WARN] [1674139974.115775685] [turtlebot4_ros_tests]: Ignoring unexpected result response. There may be more than one action server for the action '/dock' 

Test 3 actually works, beside the WARN messages, but on test 4 the turtlebot4 it performs undock and then crash.

Nodes and packages not related to navigation work fine.

Could this be a problem related to the create3 base or to the turtlebot4?
Has anyone been able to solve this problem?

JasmineRosselliSUPSI, 2023-02-01:
My navigation issues may have been fixed.
The possible cause of these problems was probably a defective SD card.
See [https://github.com/iRobotEducation/create3_docs/issues/316](url) to get full description of resolution.

roni-kreinin, 2023-02-03:
@ebear11 Did you manage to fix the issue?

If not, try using the FastDDS discovery server as mentioned [here](https://github.com/turtlebot/turtlebot4/issues/26#issuecomment-1416310217).

ebear11, 2023-02-06:
Not yet. I think our router might have been acting up on us after I realized I connected the pi to the 2.4 network and when connecting to the 5G network it failed to ping anything. If a router change does not work I will trying using that

closed 2023-06-20 by roni-kreinin