turtlebot/turtlebot4 #93: TB4 color camaera is working but streo/depth Topic is not working

rhyeonwoo, 2022-12-12:
Used 
ubuntu : 20.04
ros distro : Ros2 Galactic
robot : TB4 standard

As the title says, the camera comes out, But I cannot get stereo/depth topic well

I unplugged and re-connected the camera line, and I turned off and on the robot, but it still doesn't work.

Please check the attached photo 

[image]
[image]

What's the problem? I want to know.
Thank you.

roni-kreinin, 2022-12-13:
@rhyeonwoo Where are you having issues getting the depth image? It looks like your `ros2 topic echo` is working.

rhyeonwoo, 2022-12-15:
In the same way , In front of my desk, the other TB4 get stereo/depth topic well.
I all waved in the same way.
only this robot doesn't change the data from zero.

If it doesn't come out, is there any other way to try it?
Thank you my problem help.

closed 2023-04-14 by roni-kreinin