turtlebot/turtlebot4 #94: Appear name of RPi AP 'turtlebot4', but cannot connect WIFI with Infinity loading

potapojang, 2022-12-30:
**Please provide the following information:**
 - OS: Ubuntu 20.04
 - ROS Distro: Galactic
 - Built from source or installed:
 - Package version: 

 **Expected behaviour**
if appear name of RPi AP "Turtlebot4" and click and type passward 'Turtlebot4', so wifi will be connected -> ssh ubuntu@~.~.~.~ in terminal -> connect my PC and RPI4

 **Actual behaviour**
AP is detected, but i cannot connect RPI wifi and my PC with infinity loading. so SSH is not working
using ethernet cable dissolvent is not working with same issue. 

**To Reproduce**
Provide the steps to reproduce:
1. reboot robot and RPi - not working 
2. connect ethernet port of RPi and USB port of my PC with adapter - not working
3. connect RPi to monitor directly, and type in terminal "sudo wifi.sh -a" - pop error "vndr ie set error : -52"

**Other notes**
Add anything else you thing is important.

roni-kreinin, 2023-01-03:
If you can connect with ethernet, can you SSH in and connect the RPI to your WiFi network?

potatojang, 2023-01-04:
i can SSH with RPi's eth0 ip address (192.168.~~), but ssh ubuntu@10.42.~~ is not working

closed 2023-01-04 by potatojang