turtlebot/turtlebot4 #95: TurtleBot 4 Lite can't undock

MathisGodard, 2023-01-09:
Hello,

A customer bought the new turtlebot 4 lite, and he has some problems with the undock of the robot. 

He follows the [start guide on your git](https://turtlebot.github.io/turtlebot4-user-manual/overview/quick_start.html). He uses Thinkpad E580 with Ubuntu 20.04 LTS focal fossa to communicate with the robot. He can access to the robot via SSH, he can do "ros2 topic echo /ip" and he can "ros2 topic list"

But when he tries to undock manually, the robot is not moving : "ros2 run turtlebot4_tests ros_tests". When he tries the LED control, that is not working too. 

He doesn't have error message. 

Best regards, Mathis.

roni-kreinin, 2023-01-09:
Can you paste the output of `ros2 topic list`? It sounds like there is no communication with the Create 3 base.

MathisGodard, 2023-01-10:
Hello, Thanks for your repply.

[image]

Best regards, Mathis.

roni-kreinin, 2023-01-12:
@MathisGodard So the Create 3 topics are not showing there, only the ones used by the RPi. Has the Create 3 been connected to WiFi and configured with the same ROS_DOMAIN_ID?

roni-kreinin, 2023-01-18:
Do you have another 2.4GHz router you could test with? Or maybe just a hotspot from a phone?

MathisGodard, 2023-01-25:
I managed to make the platform work. I changed the router to only use 2.4GHz and not 2.4 and 5GHz, I guess that was the problem.

roni-kreinin, 2023-01-25:
The 2.4 and 5GHz networks may not have been bridged. There will be a new firmware update for the Create 3 soon that will allow for TB4 to operate on just the RPi WiFi. It should fix networking issues like this.

roni-kreinin, 2023-02-03:
@MathisGodard Please try using the FastDDS discovery server as mentioned [here](https://github.com/turtlebot/turtlebot4/issues/26#issuecomment-1416310217).

closed 2023-03-06 by roni-kreinin