turtlebot/turtlebot4 #97: TFs about wheels do not exist. so rviz can't show me about robot_description

potatojang, 2023-01-12:
**Please provide the following information:**
 - OS: Ubuntu 20.04
 - ROS Distro: Galactic
 - Built from source or installed:
 - Package version: (if from repository, give version from `sudo dpkg -s ros-$ROS_VERSION-turtlebot4-PACKAGE_WITH_ISSUE`, if from source, give commit hash)

 **Expected behaviour**
When i run "ros2 launch turtlebot4_viz view_robot.launch.py", rviz2 show me the perfect TF and robot_description.

 **Actual behaviour**
But i can't see perfect TF and robot_description. There is no frame about "left_wheel", "right_wheel", "wheel_drop_left", "wheel_drop_right"

what should i do...? :(

termial shows me this error
Warning: Invalid frame ID "wheel_drop_left" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.5/src/buffer_core.cpp

**To Reproduce**
Provide the steps to reproduce:
1. run something
2. launch something else
3. see the error

**Other notes**
Add anything else you thing is important.

roni-kreinin, 2023-01-12:
Does it look like this https://github.com/iRobotEducation/create3_sim/issues/125?

potatojang, 2023-01-13:
appearing errors are same with mine but it seems like wheels exist.  

in my situation, TF of wheels and wheel_drops are not exist from the beginning.
and when i run "tf2_tools view_frames.py" too

should i edit URDF?

roni-kreinin, 2023-01-13:
You shouldn't need to make any changes to the URDF. Is this coming from a robot or simulation?

potatojang, 2023-01-15:
Sorry, I don't understand your comment...

potatojang, 2023-01-15:
I didn't set about DDS. The reason why rviz can't show tf about wheels.

but i still have some problems. maybe i can fix as  refer to your comment. thank  you ;)

closed 2023-01-15 by potatojang