turtlebot/turtlebot4 #98: Turtlebot4 Configuration for Remote PC

escortKwon, 2023-01-26:
**Please provide the following information:**
 - OS: Ubuntu 20.04.5
 - ROS Distro: Galactic
 - Built from source or installed: Turtlebot 4, Turtlebot4 Desktop and Turtlebot4 Simulator
 - Package version: Turtlebot4(0.1.2), Turtlebot4 Desktop(0.1.0), Turtlebot4 Simulator(0.1.1) / All packages are cloned for latest commits on branch `galactic`

 **Expected behaviour**
Check same topics of Create3 from both Raspberry Pi 4 and Turtlebot4 Lite simultaneously

 **Actual behaviour**
 Topics from Create3 were only visible on Raspberry Pi 4 via SSH connection from Remote PC. Any topics from Create3 could not be found on local terminal of Remote PC

**To Reproduce**
Provide the steps to reproduce:
- Turtlebot4 Lite
1. Overwrite the lateset pre-installed image (0.1.3) on SD card from Raspberry Pi 4.
2. Clone the repository of `turtlebot4/turtlebot4_setup` and run `turtlebot4_setup.sh` for automatic setup.
- Remote PC
1. Clean install Ubuntu 20.04.5
2. Configure ROS2 Environment(Install ROS2 Galactic)
3. Install and build Turtlebot4 Packages: `Turtlebot4`, `Turtlebot4_Dektop`, `Turtlebot4_Simulator`
4.  Configure `~/.bashrc` as follows:
```bash
source /opt/ros/galactic/setup.bash
source ~/turtlebot4_ws/install/setup.bash
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
export CYCLONEDDS_URI='<CycloneDDS><Domain><General><DontRoute>true</></></></>'
export ROS_DOMAIN_ID=0
```
**Other notes**
I've installed and built all `Turtlebot4 packages` successfully but I've never made proper configuration between Create3 and Remote PC. Whatever I tried, I could see Create3 topics of Raspberry Pi 4 but not available for Remote PC. Also, I got some technical support directly from my sales but there were no changes. I hope there will be some better idea or methods for trouble shooting. Thanks
**P.S.** All of our school campus networks are protected by Firewall but I have tried another AP without any ethernet connection. The result was same for fail.

roni-kreinin, 2023-01-30:
Is the Create 3 connected to the same Wi-Fi network?

escortKwon, 2023-01-31:
> Is the Create 3 connected to the same Wi-Fi network?

Absolutely. Whenever I process configuring environment, I've made same connection between same Wi-Fi AP and computers

jmm-slamcore, 2023-02-01:
@escortKwon saw you also opened this ticket https://github.com/eclipse-cyclonedds/cyclonedds/issues/1476 I've also experienced issues when using cyclonedds. Switching to fastrtps seemed to work for me as a workaround.

roni-kreinin, 2023-02-03:
@escortKwon Please try using the FastDDS discovery server as mentioned [here](https://github.com/turtlebot/turtlebot4/issues/26#issuecomment-1416310217).

escortKwon, 2023-02-06:
> @escortKwon Please try using the FastDDS discovery server as mentioned [here](https://github.com/turtlebot/turtlebot4/issues/26#issuecomment-1416310217).

@roni-kreinin Thank you for providing some solutions! I've tried using the FastDDS discover server as you mentioned. I could see the same topics between Remote PC and RP4 simultaneously.

But there're some another issue. When I try undocking or moving(navigating by pressing keyboard), there come up a kind of message as follows: `Waiting for an action server to become available...`
[image]
[image]
[image]

I will be really appreciated if you could provide more troubleshooting solutions for me. But, thanks anyway for making my TB4 show same topics on both computers :)

escortKwon, 2023-02-06:
@roni-kreinin Oh, I just solved for receiving commands. I just run `ros2 daemon stop; ros2 daemon start`, It really works! Thank your for your help. It was really helpful.

closed 2023-02-06 by escortKwon